WillCodeForCats/solaredge-modbus-multi #539: Never make 2 meters/devices with same name

robchandhok, 2024-02-18:
**Describe the current behavior**
I have the weird _as-installed_ situation of my main inverter being at address 2, and the EV charger/inverter at address 1.  If I start the config at addr 2, I don't ever see the EV (obvi).  If I start at addr 1 and allow 2 inverters, I get 2 meters named M1, one of which is completely unavailable and one that is what I think it should be.

It would be more useful if they were not devices called `Solaredge M1`.  The entities that are created do get `_2` appended to remove namespace collisions, but there's no way to fiddle with the device name that I can see.

I realize my installation doesn't follow the proper modbus ID sequencing.  I note that this is being worked on (via https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/526) and thought this might be something to address at the same time.

**What should be updated or changed?**
Increment the Meter number so the second `M1` becomes `M2`, etc.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2024.2.2
 - solaredge-modbus-multi Version: 2.4.11

**Additional context**
I've been watching this integration grow in features and robustness and may, at some point, decide I could change my modbus addresses but I am also still under warranty.  I have an LG battery in the system as well which is not detected, but I don't think they enabled the battery registers and I can't get SolarEdge or the installer to do it.  Yet.

WillCodeForCats, 2024-02-18:
The meter numbers are assigned meter slots on an inverter. Any single inverter can have up to three meters attached: M1, M2, and M3. They will always be meters 1,2,3 addressed per inverter (not per global system).

You can change the device names in HA, just click on whatever the edit icon is on the device page.

WillCodeForCats, 2024-02-18:
Home Assistant appends a `_2` or `_3` to the end of the entity name if it detects a namespace collision. The integration doesn't control that. You just have to be careful to make sure all the old entities were deleted if you change around device addressing after initial setup.

As far as meter naming I explored using the inverter's ID as part of the meter name way back in [Release v2.0.0-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.0.0-pre.1).

robchandhok, 2024-02-18:
Thanks.  I get your point on the device names (I forgot that isn't just the friendly name that is changed).

I still think there's a better automatic schema for naming the meters as an enhancement.  I like being able to wipe the integration and and just re-add it (especially since you add a lot of new stuff).  So I'd rather not have to manually rename the devices every time.

I guess I'm intrigued that you decided to drop this, but don't want to waste your time explaining it to me.  Thanks.

closed 2024-02-18 by robchandhok

WillCodeForCats, 2024-02-18:
It's fine, I can revisit naming. I personally preferred seeing `I1 M1` on my system instead of just `M1`, it's just that back then it caused long term stats to be lost and that was a no-go for me as a user upgrading my own system.

WillCodeForCats, 2024-02-19:
Inverter ID added to meter and battery entity names in PR #547