WillCodeForCats/solaredge-modbus-multi #554: Plugin should accept hostnames like 'bc3b947b-modbus-proxy' as inverter IP address

schuettecarsten, 2024-02-29:
I use modbus-proxy addon for Home Assistant. This works fine but I have to use 127.0.0.1 as inverter address and expose the modbus proxy port. It would be great if your extension would accept the local addon hostname as inverter IP address.

[image]

[image]

WillCodeForCats, 2024-02-29:
I must have missed something from my other integration (tekmar-482 accepts host names like this).

WillCodeForCats, 2024-02-29:
Fixed in [Release v2.4.12-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.12-pre.4)

closed 2024-02-29 by WillCodeForCats

schuettecarsten, 2024-03-01:
Works, thank you!