WillCodeForCats/solaredge-modbus-multi #556: SolarEdge Component Serial Numbers Missing from Device Info and Meter Option/Type missing from Meter Entity Device Info

rtw819, 2024-03-03:
**Describe the feature**

1) The individual SolarEdge system components (e.g. inverters, battery unit(s), backup interface, production meters, consumption meters, etc.) are able to report serial numbers and these should be included in the module's device enumeration.  I have not found the serial numbers within the device entity descriptions or device information.  The serial number is available from the modbus details alongside the component model numbers (as "c_serialnumber" in JSON?), and could be included for visual clarity sake, especially when dealing with multiple inverters and meters.

2) The meters themselves report the type of meter they are (along with their serial number).  This information should also be included in the module's device details.  Meters show their type within the modbus details (as "c_option" in JSON?) for meter types like:
    "Export+Import"
    "Production"

3) Perhaps my understanding is off base, but I think the python solaredge_modbus module may in fact be missing on reporting of some meters (likely because the values are derived) based on what is shown by the SolareEdge Monitoring site, but that is a separate problem not withstanding the feature request herein.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version:  2024.2.0
 - solaredge-modbus-multi Version:  v2.4.11

**Additional context**
For reference, I have the following Solar Edge US system components:
SE Energy Hub 7.6kW with LG RESU10H battery and string of 30 panels
SE Energy Hub 3.8kW (follower, no battery) and string of 12 panels
SE Backup Interface (Whole Home)

The solaredge-modbus HA module (and the latest solaredge_modbus v0.8.0 python native version) report my system has 3 meters, however, the SolarEdge Monitoring Site Details page indicates 6 meters, 3 (+1) with unique serial numbers, so it's a bit confusing:

- Production Meter Type SE-RGMTR-1D-240C-B - attached to Inverter 1
- Production Meter Type SE-RGMTR-1D-240C-B - attached to Inverter 2
- Export Meter Type SE-MTR-3Y-240V-A (*) - attached to Inverter 1
- Import Meter Type SE-MTR-3Y-240V-A (*) - attached to Inverter 1
- Self Consumption Meter  (no serial, likely derived?) - attached to Inverter 1
- Consumption Meter (no serial, derived?) - attached to Inverter 1

* Export & Import Meters share the same serial number, but are listed separately in the site details

WillCodeForCats, 2024-03-03:
I don't really understand what you're asking for, the integration already reports these things.

rtw819, 2024-03-03:
No serial numbers are displayed on the device enumeration, only part model numbers?

rtw819, 2024-03-03:
`    "meters": {
        "Meter1": {
            "c_manufacturer": "SolarEdge",
            "c_model": "SE-RGMTR-1D-240C-B",
            "c_option": "Production",
            "c_version": "78",
            "c_serialnumber": "6064xxxxx",
            "c_deviceaddress": 1,
            "c_sunspec_did": 203,
            "c_sunspec_length": 105,
`

WillCodeForCats, 2024-03-03:
I still don't understand. The integration reports serial number in the device attributes.

rtw819, 2024-03-03:
I guess I must not be seeing it.   The Meter Serial Numbers and Type/Option isn't showing.  The device serial numbers aren't showing.  I see:

Device
Manufacturer
Model
Area
Integration
Battery

No serial numbers are listed?

WillCodeForCats, 2024-03-03:
None of those are "serial number" columns.

So your suggestion is to change HA front end to show a serial number column?

rtw819, 2024-03-04:
Am I missing the SolarEdge gear serial numbers somewhere else?

Perhaps serial numbers could be shown on the individual device pages?  And if possible, the reported Meter Type for Meters.

The individual Device Info pages show just about everything else, less Serial Number:

e.g.

Device info
SE7600H-USS3BBC14
by SolarEdge
Firmware: 0004.0017.0221

WillCodeForCats, 2024-03-04:
It already shows the meter type like this:
[image]

And like this:
[image]

So I still am not sure what you're asking for that isn't already in there.

If you think the serial number or hardware fields should be displayed in the device page columns, that's something you need to suggest to HA frontend. I don't control how that data is displayed by Home Assistant.

https://github.com/home-assistant/frontend

As far as serial numbers I can add that to the device properties, but again, I don't control how HA will display it.

WillCodeForCats, 2024-03-04:
[Release v2.4.13-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.13-pre.1) adds serial number to device properties.

As far as having serial numbers or other information beyond what currently shows up in the device list, it's a good suggestion to have the option to display that information in the device list, but unfortunately it's nothing I can control from the integration. That would be something HA frontend needs to do.

rtw819, 2024-03-04:
Yep, it sure is.  My bad on the meter details.  It is there in plain sight.   SolarEdge monitoring must derive the other meter values, or perhaps they are pulling from the Backup Interface somehow since I'm quite sure it has a CT meter as well.

I can't seem to find the Inverter Serial Numbers in the device details.  Though, it looks like there is already a request out there for that.

Thanks for your time and efforts!  This truly is great stuff!   Also, apologies for my confusion.

WillCodeForCats, 2024-03-04:
Serial number added to device info in PR #557.

closed 2024-03-04 by WillCodeForCats