WillCodeForCats/solaredge-modbus-multi #561: Some entities appear to never be populated

Noodleyman, 2024-03-14:
### Describe the bug

A number of entities which are anticipated to populate with values are reporting _unavailable_, but it isn't clear why.

I am using a prefix for entities, thus _number.solaredge__ becomes _number.solaredgemodbus__

Entities reporting unavailable:

- number.solaredgemodbus_i1_storage_charge_limit
- number.solaredgemodbus_i1_storage_command_timeout
- number.solaredgemodbus_i1_storage_discharge_limit
- number.solaredgemodbus_i1_ac_charge_limit

Other entities from the same device do appear to be collecting data, for example:
- number.solaredgemodbus_i1_backup_reserve = 4
- number.solaredgemodbus_i1_site_limit = 3668

Per the template design notes, 
_When an inverter goes offline (returns an error, incomplete read, timeout, or other errors) entities will change state to unavailable so that stale or bad data is not reported:_

It is assumed there is an error but it isn't clear what that error is at the moment. It is also assumed the values should populate at least once since the module was enabled (this may be me being incorrect).

### Expected behavior

Values other than undefined to be available

### Screenshots

[image]

Battery is detected:
[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-e618d9b4098f6cae5cc6371519c59951.json](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14600765/config_entry-solaredge_modbus_multi-e618d9b4098f6cae5cc6371519c59951.json)

### Home Assistant Version

11.5

### solaredge-modbus-multi Version

2.4.12

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

### Additional Context

Attempting to install Predbat for SE, as described here:
https://github.com/springfall2008/batpred/issues/181

Noodleyman, 2024-03-14:
I did try to share a debug log, but it reported as too long to attach. 

However, debug logs do report the data which the entities seem to not reflect:

```
2024-03-14 10:02:21.175 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: ac_charge_limit 0x0 <class 'float'>
2024-03-14 10:02:21.176 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: discharge_limit 0x459c4000 <class 'float'>
2024-03-14 10:02:21.176 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: charge_limit 0x459c4000 <class 'float'>
2024-03-14 10:02:21.176 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_timeout 0xe10 <class 'int'>
```

Noodleyman, 2024-03-14:
anddddddddddddd, closing. I figured it out by checking into the code.  for the properties of interest to be populated the solaredge_i1_storage_control_mode value should be set to Remote Control.

closed 2024-03-14 by Noodleyman

Noodleyman, 2024-03-14:
noted. thanks for the update. I was trying to keep things "as they are" until I am ready to cut over, which caught me off guard

WillCodeForCats, 2024-03-14:
I can understand that "unavailable" is not clear when it could be either wrong mode or communication failure, but HA doesn't have any other entity properties I could use to show the difference.