WillCodeForCats/solaredge-modbus-multi #562: Connection errors

cburghardt, 2024-03-16:
### Describe the bug

The connection to the inverter dies roughly once per day. It doesn't recover so I need to reload the integration.
I think this started with 2024.3.1

### Expected behavior

Connection recovers

### Diagnostic File

[config_entry-solaredge_modbus_multi-b86bc24f1ca70258f7ffd41889955498.json](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14622754/config_entry-solaredge_modbus_multi-b86bc24f1ca70258f7ffd41889955498.json)

### Debug logs

```shell
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solaredge_modbus_multi
Quelle: helpers/update_coordinator.py:318
Integration: SolarEdge Modbus Multi (Dokumentation, Probleme)
Erstmals aufgetreten: 15. März 2024 um 20:16:20 (326 Vorkommnisse)
Zuletzt protokolliert: 08:05:52

Unexpected error fetching SolarEdge Coordinator data: AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 318, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 191, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 228, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 225, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 364, in async_refresh_modbus_data
    await inverter.read_modbus_data()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 826, in read_modbus_data
    inverter_data = await self.hub.modbus_read_holding_registers(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 452, in modbus_read_holding_registers
    result = await self._client.read_holding_registers(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 179, in async_execute
    self.close(reconnect=True)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/tcp.py", line 92, in close
    super().close(reconnect=reconnect)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 130, in close
    self.connection_lost(asyncio.TimeoutError("Server not responding"))
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 288, in connection_lost
    self.close(intern=True)
TypeError: AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'
```

### Home Assistant Version

2024.3.1

### solaredge-modbus-multi Version

2.4.12

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-03-16:
Try asking in pymodbus.

https://github.com/pymodbus-dev/pymodbus/pull/2080

cburghardt, 2024-03-16:
So is that a regression? Can I change the config somehow?
If that's an issue of the underlying module it would affect everybody?

WillCodeForCats, 2024-03-16:
I can try pushing to pymodbus 3.6.6 in a pre-release.

WillCodeForCats, 2024-03-17:
It's probably a regression, definitely a bug in pymodbus. Updating HA can update the version of pymodbus that HA ships with. My integration will use whatever the current version is as long as it meets the minimum API requirements (currently pymodbus 3.5.x). I will usually push releases every now and then to track some HA equivalent minimum.

You can change the pymodbus version the integration will require by editing `manifest.json` file.

WillCodeForCats, 2024-03-17:
So, for example, if you weren't seeing this problem in HA 2024.2.x but the problem started in 2024.3.1 or 2024.3.0 then you can determine which version of pymodbus is causing it based on what each those versions of HA ships with.

cburghardt, 2024-03-17:
I'll try the version with pymodbus 3.6.6 once it is out. For now I am reloading the integration every few hours.

ionionascu, 2024-03-17:
Just for the record, I can confirm that setting `"requirements": ["pymodbus==3.6.4"]` in `manifest.json` fixes the issue after a restart. I am running `OS Version: Home Assistant OS 12.1` and `Home Assistant Core: 2024.3.1` and I only noticed the issue in `2024.3.1` where the connection to the inverter terminates after only a few minutes following a restart.

WillCodeForCats, 2024-03-17:
HA 2024.3.1 comes with pymodbus 3.6.5

HA 2024.3.0 comes with pymodbus 3.6.4

All of the HA 2024.2.x releases had pymodbus 3.6.3

thargy, 2024-03-20:
As per [this comment](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/535#issuecomment-2010643898), I have manually modified my `manifest.json` `requirements` to `3.6.6` as below:
```json
{
  "domain": "solaredge_modbus_multi",
  "name": "SolarEdge Modbus Multi",
  "codeowners": ["@WillCodeForCats"],
  "config_flow": true,
  "dependencies": [],
  "documentation": "https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki",
  "integration_type": "hub",
  "iot_class": "local_polling",
  "issue_tracker": "https://github.com/WillCodeForCats/solaredge-modbus-multi/issues",
  "loggers": ["custom_components.solaredge_modbus_multi"],
  "requirements": ["pymodbus>=3.6.6"],
  "version": "2.4.13-pre.3"
}
```

Core
2024.3.1
Supervisor
2024.03.0
Operating System
12.1
Frontend
20240307.0

I will now monitor for further issues.

WillCodeForCats, 2024-03-20:
Basically 2.4.13-pre.3 just aligns requirements with what Home Assistant 2024.3.0 uses. There are no code changes in it.

thargy, 2024-03-20:
> Basically 2.4.13-pre.3 just aligns requirements with what Home Assistant 2024.3.0 uses. There are no code changes in it.

My system rebooted with the higher requirement (`>=3.6.6` vs `>=3.6.4`), and is currently working correctly. Hopefully, the connection issue will disappear, at which point you may consider increasing the requirement more permanently. I will update if nothing happens in the next week.

WillCodeForCats, 2024-03-20:
Yes if that fixes it I will push the pymodbus min to 3.6.6.

cyberfilou, 2024-03-20:
hello I have the haos version so no command line access to add pymodbus higher version or there is a method thank you

thargy, 2024-03-20:
> hello I have the haos version so no command line access to add pymodbus higher version or there is a method thank you

If you are comfortable editing a file, you can use your favourite file viewer addon (I use the VIsual Studio Code addon in HAOS), and navigate to `custom_components/solaredge_modbus_multi/manifest.json` in your `config` folder.

You will see a file that looks like above, the line
`"requirements": ["pymodbus>=3.6.4"],`
needs to be changed to
`"requirements": ["pymodbus>=3.6.6"],`

And then you should restart Home Assistant and confirm the addon loads correctly.

Note, manual changes to the manifest will be overridden next time you update the addon, so this is a temporary change.

WillCodeForCats, 2024-03-22:
Home Assistant [2024.3.2](https://github.com/home-assistant/core/releases/tag/2024.3.2) now has pymodbus v3.6.6.

cyberfilou, 2024-03-22:
> > bonjour j'ai la version haos donc pas d'accès en ligne de commande pour ajouter une version supérieure de pymodbus ou il y a une méthode merci
> 
> Si vous êtes à l'aise avec l'édition d'un fichier, vous pouvez utiliser votre module complémentaire de visualisation de fichiers préféré (j'utilise le module complémentaire VIsual Studio Code dans HAOS) et accéder à `custom_components/solaredge_modbus_multi/manifest.json`votre `config`dossier.
> 
> Vous verrez un fichier qui ressemble à celui ci-dessus, la ligne `"requirements": ["pymodbus>=3.6.4"],` doit être modifiée en `"requirements": ["pymodbus>=3.6.6"],`
> 
> Et puis vous devez redémarrer Home Assistant et confirmer que le module complémentaire se charge correctement.
> 
> Notez que les modifications manuelles apportées au manifeste seront annulées la prochaine fois que vous mettrez à jour le module complémentaire, il s'agit donc d'une modification temporaire.

Thank you
the modification are install

WillCodeForCats, 2024-03-22:
Fixed in Home Assistant 2024.3.2

closed 2024-03-22 by WillCodeForCats