WillCodeForCats/solaredge-modbus-multi #565: EV Charger support

WillCodeForCats, 2024-03-28:
### Discussed in https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/564

<div type='discussions-op-text'>

<sup>Originally posted by **mikedeluca** March 27, 2024</sup>
So I was digging around to try to figure out why some sites batteries work and at others they don't show up and I ran across a Modbus address that wasn't one of the 3 inverters. 

{'c_id': 'SunS', 'c_did': 1, 'c_length': 65, 'c_manufacturer': 'SolarEdge', 'c_model': 'SE-EV-SA-US-40P', 'c_version': '0004.0019.0036', 'c_serialnumber': '63002925', 'c_deviceaddress': 4, 'c_sunspec_did': 103, 'c_sunspec_length': 50, 'current': 0, 'l1_current': 0, 'l2_current': 0, 'l3_current': 0, 'current_scale': -32768, 'l1_voltage': 0, 'l2_voltage': 0, 'l3_voltage': 0, 'l1n_voltage': 0, 'l2n_voltage': 0, 'l3n_voltage': 0, 'voltage_scale': -32768, 'power_ac': -32768, 'power_ac_scale': -32768, 'frequency': 0, 'frequency_scale': -32768, 'power_apparent': -32768, 'power_apparent_scale': -32768, 'power_reactive': -32768, 'power_reactive_scale': -32768, 'power_factor': -32768, 'power_factor_scale': -32768, 'energy_total': 0, 'energy_total_scale': -32768, 'current_dc': 0, 'current_dc_scale': -32768, 'voltage_dc': 0, 'voltage_dc_scale': -32768, 'power_dc': -32768, 'power_dc_scale': -32768, 'temperature': -32768, 'temperature_scale': -32768, 'status': 1, 'vendor_status': 0, 'rrcr_state': 0, 'active_power_limit': 100, 'cosphi': 0, 'commit_power_control_settings': 0, 'restore_power_control_default_settings': 0, 'reactive_power_config': 0, 'reactive_power_response_time': 0, 'advanced_power_control_enable': 0, 'export_control_mode': 0, 'export_control_limit_mode': 0, 'export_control_site_limit': 0}

This is ID 4 on the Modbus TCP. The model and serial match my Solaredge EV Charger. Feels like this would be interesting functionality to add... I'll try to do some more scanning of registers</div>

WillCodeForCats, 2024-03-29:
EV charger appears to present as an inverter based on the sunspec ident block.

deqw, 2024-03-31:
Looking at the model it is a Keba P40 charger disguised as a SolarEdge Inverter. Maybe just for presentation purpose?

WillCodeForCats, 2024-04-02:
Maybe making it look like an inverter would let it reuse modbus definitions without SolarEdge having to support new types, but also if it looks like an inverter it wouldn't be possible to detect except by trying to match model strings.

If some of those values change when the charger is charging I could make a restricted display for a charger class.

deqw, 2024-04-03:
I have been using this [Keba integration](https://github.com/dannerph/homeassistant-keba) for a while to control the SolarEdge EV charger but stopped using it. Now using Node-Red on Home Assistant for this purpose which gives me more control.

WillCodeForCats, 2024-04-03:
I can find modbus data about Keba P30 but not about P40.

If SolarEdge is just using the P40 rebranded and the P30 modbus still works with the P40, I can try that.

WillCodeForCats, 2024-04-03:
I can't use the UDP guide unfortunately, it needs to be modbus through the SolarEdge inverter bus to work with this integration.

mikedeluca, 2024-04-03:
The solaredge charger in this case does not apear to be the Keba. Solaredge seems to have their own EVSE for the North American (and I think Australian) markets. https://www.solaredge.com/us/products/residential/solaredge-home-ev-charger. The Keba is this one: https://www.solaredge.com/en/products/residential/ev-charging. I'll try to get some info from it with the car connected and charging. Its technically my secondary EVSE since I primarily use a Tesla wall connector, so I need to move some things to get the car close enough. I suspect your correct, that they just recycled sunspec registers to make things simple. I'll also see what i can pull from setapp.

deqw, 2024-04-03:
@mikedeluca the Keba one you reference is a P30, not a P40.

mikedeluca, 2024-04-03:
So confirmed in setapp its configured as a solaredge follower, just like an inverter. Also was able to confirm in setapp that it contains FCC ID 2AGPT-PLNX, which is a solaredge comms board. https://fccid.io/2AGPT-PLNX/User-Manual/Users-Manual-rev-pdf-3941269. One interesting thing, the label on the outside proclaims it as a SE-EV-SA-US-40N not a 40P as the modbus returns. probably doesnt matter, but for future readers. Full serial is SJ5121-063002925-B1. In Setapp the serial is SN 05F5EB22-07 in the status page (probably the comms board?) and 63002925-B1 in the information screen. Im running Version 4.20.32/2.2.12/4.20.64 (CPU/EVcharger/WSA). I'm not clear how often these are configured via Modbus vs Wifi or ethernet. Mine is setup that way and is connected directly to inverter #1 for both modbus and AC power.

mikedeluca, 2024-04-04:
[Debuglog.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14860888/Debuglog.txt)

This is the debug log output from a scan of a bunch of the registers. last few scans are on the EVSE wile charging

bisscuitt, 2024-06-07:
I believe I have one of these inverters. It's an SE5000H with integrated EV Charger (Australia) and it would be awesome if I could monitor and control charging over modbus.

Can you give me some advice on how to provide required modbus dumps ?

laurencei, 2024-08-08:
@WillCodeForCats  - do you have an EV charger? 

Not sure if your seen this - but when adding the SolarEdge EV charger - some extra options appear in the SolarEdge app

![IMG_3204](https://github.com/user-attachments/assets/b3e5a324-6c6a-4250-b45a-993585846f

[image]
c2)

Specifically the part I'm most interested in is been able to toggle the "excess solar" on/off, and the "charge now" override.

Is there something I can do to help?

laurencei, 2024-08-08:
> I'm not clear how often these are configured via Modbus vs Wifi or ethernet. Mine is setup that way and is connected directly to inverter 1 for both modbus and AC power.

That is how mine has been installed as well.

laurencei, 2024-08-08:
I found this guide for the SolarEdge EV charger. Has some specifics on error codes etc that might be useful: https://knowledge-center.solaredge.com/sites/kc/files/smart-ev-charger-installation-guide-na.pdf

WillCodeForCats, 2024-08-08:
No, I do not have an EV charger on my system.

There are various modbus scanning utilities that might be able to help. This one is cross platform: https://github.com/SciFiDryer/ModbusMechanic

laurencei, 2024-08-22:
I've actually found an issue. When running SolarEdge in "Remote Control" mode, the EV charger doesnt work correctly, specifically the "pending excess solar" option doesnt trigger. I can only trigger the EV charger manually via the app on or off.

I'm going to reach out to SolarEdge about it - might be a way to get some modbus data on how to control the EV charger ourselves, since the app seems to lose the ability when we do remote control.

(Solaredge specifically gave me remote control back, so I can press them on helping to fix this issue).

deqw, 2024-08-22:
Charging on excess solar didn't work for me either. Since two weeks it is working however after an inverter (SE10K-RWB48BFN4) firmware update (0004.0021.0019). And still working after 0004.0021.0021 that it is on now. Load balancing
is also working ok. Before I - could/had to - trigger charging eg. by closing the car again (did not matter if it was open or closed already). Maybe a direction for them to look into.

laurencei, 2024-08-28:
Yeah - they responded they have updated the EV Charger firmware - but I cant see it, so I dont know what exactly changed.

Will check how it goes.

d0minander, 2024-09-23:
Regarding the models SolarEdge actually has two models, one for single phase and one for three phase.
The Keba P30 version is the three phase, it even runs on the same firmware, but seems to be one version behind the official Keba version.

Zandor300, 2024-09-25:
Just want to point out that EV's have a policy, at least in the European Union, that when the max current the EVSE provides to the EV falls below 6A, it should stop charging altogether.

Assuming 230V/400V grid:

- If you have a three phase installation and an EV that supports three phase charging, you need at least ~4,2 kW of excess PV before your EV will start charging.
- If the charger or EV only supports single phase, it will start charging at ~1,4 kW of excess PV.

So for people who wonder why excess PV charging isn't working, this is probably why. If you have that enabled and you hit the "start charging" button in the mySolarEdge app, you will force the charger to start charging at full power, throwing the excess PV charging setting out the window for the current session.

SolarEdge doesn't provide the option to configure the P30 to start charging on single phase if your car supports three phase and you only have 1,4 kW of excess PV. Maybe the onboard settings UI of the P30 has this option available?

d0minander, 2024-09-25:
@Zandor300 Good points.
When using single phase charging, can the inverter then boost power to the phase that the EV charger is using?
Becouse if the inverter has 3kw of excess that would only be 1kw per phase?

I could access the WebUI of the charger and it seems to have all the options that the Keba unit has, so if the Keba unit supports it, the Solaredge one should, but i dont have access to it right now since its installed at another property.
https://www.keba.com/en/emobility/products/x-series/x-series

Zandor300, 2024-09-25:
Only the 3phase Home Hub inverter can provide different amounts of power to the phases. The rest of the 3phase inverters split the power output evenly over the 3phases.

So when you charge with single phase 3 kW, the 3phase inverter will provide 1kW on each phase. In the Netherlands, the smart meter provided by the energy company will perform net metering between the phases and you only have to pay for the sum of electricity used. So when you export 1 kW on phase 1 and import 1 kW on phase 2, you don't pay anything.  
*(This is different from the net metering where you can export 30 kWh during the day and import again in the evening)*

Don't know about other countries.

d0minander, 2024-09-25:
Ok thanks for the info.
I dont know how our net metering in Sweden works, will look in to it.
Seems the new Keba P40 has built in phase switching, but the P30 that SolarEdge uses needs an external box for phase switching.

bisscuitt, 2024-10-22:
> There are various modbus scanning utilities that might be able to help. This one is cross platform: https://github.com/SciFiDryer/ModbusMechanic

@WillCodeForCats  I have the SE5000H inverter with built in EV charger

I'm trying to scan using ModbusMechanic and https://github.com/sanny32/OpenModScan but not having luck getting any registers to show any data.

Can you provide a quick howto on how to make these devices show any data?

Home Assistant is able to read the data with the same connection parameters I am using in the above scanner apps. I've tried a scan from 0 through to 65535 on the registers but everything comes back as a zero value.

I've tried multiple values for Device ID (1,2,3,4,101,102,103) but only seeing zero value responses. I can confirm that the TCP connection to the inverter is working as expected, I just can't get the right parameters to return any data.

If I can get the readings, I can provide some dumps and help work out what registers provide details for the EV charger. Thanks.

Zandor300, 2024-10-22:
@bisscuitt There are input, coil and holding registers. As far as I know, SolarEdge only uses holding registers. Possibly you used functions for reading input or coil registers?

d0minander, 2024-10-23:
I can also help out doing a modbus scan in someone can point me to a good scan tool for MacOS.
I have an SE17K with the three phase EV charger.

jayjay0010, 2025-02-08:
Hi @VLWilliam 
I have as well a solaredge EV charger (in fact a Keba P30 rebranded) and I would be really interested on you work to integrate it via modbus (I currently use the keba integration but I am not able to enable the charge because it is directly canceled by solaredge backend some how). 

Thanks

d0minander, 2025-02-13:
I read somewhere in one of the charger documentations that the charger is limited to one modbus connection/host.
So if you have integrated the charger with your inverter you are already using it, im guessing that the modbus connection from home assistant then gets ignored/overided?
It also mentions its only possible to use either Modbus TCP or UDP, not both, do we know what the inverter integration is using?

It would probably be best if this SolarEdge integration was used for the charger communication via the inverter, if thats possible, so their is only one modbus connection with the charger.

jayjay0010, 2025-02-16:
@VLWilliam @d0minander 
My solaredge charger is connected via OCCP to solaredge "backend" (csms.solaredge.com). I don't think there is a direct communication between my inverter and the charger via modbus, even if modbus looks activated on my charger

d0minander, 2025-02-16:
> [@VLWilliam](https://github.com/VLWilliam) [@d0minander](https://github.com/d0minander) My solaredge charger is connected via OCCP to solaredge "backend" (csms.solaredge.com). I don't think there is [...]

Yes the charger is connected to the OCCP backend, but also (if installed properly) with the inverter, thats why you enable the DIP switch for modbus on the charger as part of the install, and with the software configure the connection on the inverter.
So if this has not been done for you, you are missing part of the integration.
Se the guide:
https://www.youtube.com/watch?v=RIyGuozESM4&ab_channel=SolarEdgePV

joelegg, 2025-05-02:
I have my RV charger set up and ready to go and it's connected with rs485. Do I need to use a development branch in order to get this working as the integration is not detecting it?. Thank you

WillCodeForCats, 2025-05-04:
The integration can't detect the charger becasue it's not supported.

If you know the unit ID you can manually configure it and others have reported it appears as an inverter, but someone needs to scan modbus registers while cycling the inverter through its various modes to find out which things change and how it appears different from known devices.

Or someone needs to send me a charger so I can try to decode it myself.

joelegg, 2025-05-04:
i've tried to add it with its direct ip address and unit id, no go - solaredge said i first need to charge my car before everything sets up in the app which is odd. 

ill play around with it as soon as i can to see what i can see.

d0minander, 2025-05-05:
As mentioned before, the three phase SE charger is a rebranded Keba P30, it follows the same software updates also.
There is a modbus guide for the P30, so my guess would be that this is what the inverter uses to talk to the charger.
https://www.keba.com/download/x/dea7ae6b84/kecontactp30modbustcp_pgen.pdf

Maybe that can be of some help?
The Keba also alows to extract logs from the charger, would it be of use if i try to extract the modbus logs?
//Niklas

joelegg, 2025-05-13:
According to SolarEdge support, my current EV charger (the one pictured here: https://www.solaredge.com/aus/products/residential/ev-charging) does not support Modbus communication and only uses the SolarEdge protocol. While I can charge using solar power, I’d prefer to automate the process myself.

Can anyone recommend a reliable EV charger that integrates well with SolarEdge and supports Home Assistant for custom control?

I’m really happy with the existing SolarEdge integration for managing my system, thank you for that!

d0minander, 2025-09-19:
Bumping this, if there is anything i can do to help with this enhancement, like scanning my inverter or ev-charger i am happy to help, just let me know what you need.

briadelour, 2025-12-09:
I made a little progress.  For my SolarEdge 40P that uses an RS485 serial connection to the inverter appears to be reporting power on block 69 register 76 that can be used to determine when the charging is idle vs in a charging session.  This what it looked like when I was charging from the SolarEdge dashboard, stopped charging and then started it up again.  Pretty close to real-time.  The kW must include some overhead because my EV charger usually gets up to 9.4-9.6kW.

`======================================================================
EVSE BLOCK 69 - 2025-12-08 20:38:22
Power Register 76: 2429 -> 1.65 kW | Status: Charging
======================================================================
======================================================================
EVSE BLOCK 69 - 2025-12-08 20:38:27
Power Register 76: 2431 -> 0.84 kW | Status: Charging
======================================================================
======================================================================
EVSE BLOCK 69 - 2025-12-08 20:38:32
Power Register 76: 2433 -> 0.02 kW | Status: Idle
======================================================================
======================================================================
EVSE BLOCK 69 - 2025-12-08 20:38:37
Power Register 76: 2433 -> 0.02 kW | Status: Idle
======================================================================
======================================================================
EVSE BLOCK 69 - 2025-12-08 20:38:42
Power Register 76: 2434 -> -0.38 kW | Status: Idle
======================================================================
======================================================================
EVSE BLOCK 69 - 2025-12-08 20:38:47
Power Register 76: 2433 -> 0.02 kW | Status: Idle
======================================================================
======================================================================
EVSE BLOCK 69 - 2025-12-08 20:38:52
Power Register 76: 2407 -> 10.59 kW | Status: Charging
======================================================================
======================================================================
EVSE BLOCK 69 - 2025-12-08 20:38:57
Power Register 76: 2399 -> 13.85 kW | Status: Charging
======================================================================
======================================================================
EVSE BLOCK 69 - 2025-12-08 20:39:02
Power Register 76: 2400 -> 13.44 kW | Status: Charging
======================================================================`

[evse_block_69_reg_76_poll.py](https://github.com/user-attachments/files/24044234/evse_block_69_reg_76_poll.py)

WillCodeForCats, 2025-12-10:
Sunspec often uses scaling registers to calculate values, so there is probably an additional value that needs to be applied to calculate power.

briadelour, 2025-12-10:
The US 40P model does also have the option to connect via IP, but I chose the serial connection to the inverter option.  This allows me to avoid having to use a dedicated electrical breaker for the EV Charger because instead it is connected to the serial bus bar on the inverter.  This does mean that the EV Charger is running at 32A instead of 40A, but with this setup I still push around 9.4-9.6 kW max power on a charge.  I am able to take advantage of a feature SolarEdge calls "solar boost mode" which allows me to use both PV power and Grid to charge while still also having the option to charge only when I have excess solar.  Given the electric company is trying to do away with my 1:1 credit for net metering from solar, in the future I'm hoping to plug an EV in to use the excess solar when the power company plans to credit me 20% of what they are offer today unless I get battery backup sooner.

briadelour, 2025-12-11:
Here's more detail:

At 6:40 it was Plugged In - Pending Automation according to the SE app.
At 6:42 I started Charging and it hit 9.4 kW
At 6:47 I stopped it
At 6:49 I started charging again

[image]

It's morning and still dark out so not producing.  If it get sunnier I'll try to run the same test to compare the numbers when I'm hybrid charging.

WillCodeForCats, 2025-12-11:
If that is voltage then that's where the scaling factor comes in: an unsigned int value of 2435 can be the value 243.5 with a -1 scale factor value from a nearby register. The scale range should be -10 through 10.

briadelour, 2025-12-12:
M2 AC Power is likely not going to work reliably.  I realize now that this must be the imported energy meter for the entire house.  Running the clothes dryer drives this up over -8,000 W.  Same problem when I'm not charging and running the clothes dryer, I see spikes when watching register 76.  

So I think we are back to square one in finding some other register dedicated to reading the EV Charger if there is one.

briadelour, 2025-12-23:
Made some progress this evening.  Here are my notes for a reverse engineered Register Map for the SolarEdge EV charger hardwired to the inverter.

### Assumptions: 

- SolarEdge EV Charger with solar boost kit is hardwired to Inverter.  
- Inverter slave ID 1.
- EV Charger is identified as device ID 3.

### Discovered Static Registers:
**Device ID 3**
40004 - Manufacturer, returns SolarEdge
40020 - Charger Model, returns SE-EV-SA-US-40P
40044 - Firmware Version, returns 0004.0018.0032
40052 - Serial Number, returns 8 digit serial number

**Device ID 1**
378 - EVSE Model ID, returns 6000
379 - Model Length, returns 65534
632 - Frequency Hz (divide by 100), returns 6000
633 - Frequency Hz (divide by 100), returns 6000
638 - Voltage (/10), returns 2400
797 - Possible Max power with overheads in W, returns 12000
847 - Possible Max power in W, returns 11000
849 - Possible Max power in W, returns 11000
875 - Ground Fault threshold in mA, returns 5
982 - Ground Fault threshold in mA, returns 5
987 - Ground Fault threshold in mA, returns 5
988 - Possible Max power with overheads in W, returns 12000
991 - Possible Max power in W, returns 11000
1359 - Hardwired Amperage limit from Grid, returns 32
40362 - Possible SunSpec EVSE model, returns 203
384 - Charging Indicator, 0 = Charging, 65535 = Not Charging
394 - Cable Detection, <65530 = Plugged in, 65531+ = Unplugged
396 - Status word, secondard status indicator, ~56,000 range = Unplugged
411 - Possible charging rate, accumulates but may need some type of scaling as it often reads too high

### Others test (many had noise):
389 - Not reliable as charging state, alternates between 0 and 65535, possibly time of day dependent working better at night
390-393 - too noisy to use, appeared to be a 16-bit status word, Bit 12 = Charging, Bit 10 = Cable Connected, Bit 9 Power available, but not consistent
380 - During middle of day coincided with solar production, oscillates between ~3,006 to ~62,000
	- During evening oscillations are between 50,000 - 62,000
	- When household wakes up oscillations are between 45,000 - 57,000
385 - appears to be some charging meter but very noisy
	- when not charging returns values between 8,000 - 15,000
	- when charging appears to jump above 15,000 - 30,000 consistently
	- when not EV Charging graph appears to match the inverse of I1 M2 AC Power some
386 - could be a charging power monitor with some indicator to calculate power when charging
387 - same as 386
397 - could be AC current, oscillates -3,000 to -5,500 at points, then +900 to 1,100
480 - remains 0 when PV system is not producing (no solar power), values between 0 - 634
	- bell curve tracks well with solar production hours
	- wondering if this might be the solar boost kit reading that determines how much solar is available to charger
408 and 409 (uint32)- Timestamp?
413 - Not sure what this is, values jump around from 1,184 - 65,320

Here's an example of the dashboard I used to narrow down the values.  I tried to see if I could use Claude or ChatGPT to try to find a connection with the bit values and a possible keyword, but I couldn't find anything that worked more reliably then just finding the threshold and basing the sensor value off of that.
[image]

briadelour, 2025-12-23:
@WillCodeForCats  Turns out this might be a lot easier than the approach I was taking.  I think SolarEdge does conform to SunSpec Standards for EV Chargers, I was just reading the wrong registers.  I think that 40188 - 40292 has everything that's needed to align with SunSpec model 203.  I'll type up my notes.  Getting this added to the solaredge-modbus-multi would be great to have all the modbus registers efficiently managed.

Still need to validate with more testing.

[basicrequiredforSunSpec203onSE40P.txt](https://github.com/user-attachments/files/24318074/basicrequiredforSunSpec203onSE40P.txt)

[image]

briadelour, 2025-12-24:
I'm going to stop trying to reverse engineer this for now.  With each attempt I can't seem to make progress for more than a couple hours, before the monitoring I set up starts throwing false positives.  I think if SolarEdge does have modbus registers that allow the ability to distinguish the EV charger from other household demand, I haven't found them.  What they use on their SolarEdge dashboard must be some proprietary way of monitoring.  I did open a customer support cases asking for this just as I did the public APIs to see if they are willing to share.  The SE platform has been great with HA, this would be huge bonus for me.  I even tried Claude/ChatGPT by feeding them json output from python scripts, screenshots and activity log data from HA to help me analyze for some type of pattern, but each conversation leads to disappointment.  With the information I've found, the only way I can see doing this is by building an adjustable baseline with varying thresholds for differential metering based on time of day/year, solar status and general load.  Otherwise, the counters and register values can't seems to highlight when you are pushing 9.4-9.6 kW to an EV.

WillCodeForCats, 2025-12-24:
@briadelour Can you try downloading scan.py from here: https://github.com/thargy/modbus-scanner

And run it against your solaredge system with the following arguments:

`scan.py -d 1-32 -x 100 ip_address`

I would like to see how it identifies itself compared to the inverters.

briadelour, 2025-12-24:
Scanning ID: 1 ... INVERTER Received (27 bytes): 00 01 00 00 00 15 01 03 12 53 75 6e 53 00 01 00 41 53 6f 6c 61 72 45 64 67 65 20

(only found one out of 32)

WillCodeForCats, 2025-12-24:
if it shows any unknown response I want to see those too.

briadelour, 2025-12-24:
That's interesting, trying a second time it found what I think is the EV Charger!

Scanning ID: 1 ... INVERTER Received (27 bytes): 00 01 00 00 00 15 01 03 12 53 75 6e 53 00 01 00 41 53 6f 6c 61 72 45 64 67 65 20
Scanning ID: 2 ... Timedout after 0.5s
Scanning ID: 1 ... INVERTER Received (27 bytes): 00 01 00 00 00 15 01 03 12 53 75 6e 53 00 01 00 41 53 6f 6c 61 72 45 64 67 65 20
Scanning ID: 1 ... INVERTER Received (27 bytes): 00 01 00 00 00 15 01 03 12 53 75 6e 53 00 01 00 
Scanning ID: 1 ... INVERTER Received (27 bytes): 00 01 00 00 00 15 01 03 12 53 75 6e 53 00 01 00 41 53 6f 6c 61 72 45 64 67 65 20
Scanning ID: 2 ... Timedout after 0.5s
Scanning ID: 3 ... INVERTER Received (27 bytes): 00 03 00 00 00 15 03 03 12 53 75 6e 53 00 01 00 41 53 6f 6c 61 72 45 64 67 65 20
Scanning ID: 4 ... Timedout after 0.5s
Scanning ID: 2 ... Timedout after 3.0s
Scanning ID: 4 ... Timedout after 3.0s
Scanning ID: 5 ... Timedout after 0.5s
Scanning ID: 6 ... Timedout after 0.5s
Scanning ID: 7 ... Timedout after 0.5s
Scanning ID: 8 ... Timedout after 0.5s
Scanning ID: 5 ... Timedout after 3.0s
Scanning ID: 6 ... Timedout after 3.0s
Scanning ID: 7 ... Timedout after 3.0s
Scanning ID: 8 ... Timedout after 3.0s
Scanning ID: 9 ... Timedout after 0.5s
Scanning ID: 10 ... Timedout after 0.5s
Scanning ID: 11 ... Timedout after 0.5s
Scanning ID: 12 ... Timedout after 0.5s
Scanning ID: 9 ... Timedout after 3.0s
Scanning ID: 10 ... Timedout after 3.0s
Scanning ID: 11 ... Timedout after 3.0s
Scanning ID: 12 ... Timedout after 3.0s
Scanning ID: 13 ... Timedout after 0.5s
Scanning ID: 14 ... Timedout after 0.5s
Scanning ID: 15 ... Timedout after 0.5s
Scanning ID: 16 ... Timedout after 0.5s
Scanning ID: 13 ... Timedout after 3.0s
Scanning ID: 14 ... Timedout after 3.0s
Scanning ID: 15 ... Timedout after 3.0s
Scanning ID: 16 ... Timedout after 3.0s
Scanning ID: 17 ... Timedout after 0.5s
Scanning ID: 18 ... Timedout after 0.5s
Scanning ID: 19 ... Timedout after 0.5s
Scanning ID: 20 ... Timedout after 0.5s
Scanning ID: 17 ... Timedout after 3.0s
Scanning ID: 18 ... Timedout after 3.0s
Scanning ID: 19 ... Timedout after 3.0s
Scanning ID: 20 ... Timedout after 3.0s
Scanning ID: 21 ... Timedout after 0.5s
Scanning ID: 22 ... Timedout after 0.5s
Scanning ID: 23 ... Timedout after 0.5s
Scanning ID: 24 ... Timedout after 0.5s
Scanning ID: 21 ... Timedout after 3.0s
Scanning ID: 22 ... Timedout after 3.0s
Scanning ID: 23 ... Timedout after 3.0s
Scanning ID: 24 ... Timedout after 3.0s
Scanning ID: 25 ... Timedout after 0.5s
Scanning ID: 26 ... Timedout after 0.5s
Scanning ID: 27 ... Timedout after 0.5s
Scanning ID: 28 ... Timedout after 0.5s
Scanning ID: 25 ... Timedout after 3.0s
Scanning ID: 26 ... Timedout after 3.0s
Scanning ID: 27 ... Timedout after 3.0s
Scanning ID: 28 ... Timedout after 3.0s
Scanning ID: 29 ... Timedout after 0.5s
Scanning ID: 30 ... Timedout after 0.5s
Scanning ID: 31 ... Timedout after 0.5s
Scanning ID: 32 ... Timedout after 0.5s
Scanning ID: 29 ... Timedout after 3.0s
Scanning ID: 30 ... Timedout after 3.0s
Scanning ID: 31 ... Timedout after 3.0s
Scanning ID: 32 ... Timedout after 3.0s
Closing connection ... SUCCEEDED

Finished scan! Found 2 inverters.

[image]

[image]

briadelour, 2025-12-24:
I did not know the integration could handle the same IP for two different device IDs!

After doing a scan in three states and putting it through claude for a comparison and analysis this is what I got.  
[register_analysis_summary.txt](https://github.com/user-attachments/files/24333864/register_analysis_summary.txt)
[modbus_comparison.txt](https://github.com/user-attachments/files/24333863/modbus_comparison.txt)

Will keep running tests, but only on device ID 3.

I'll try to do some continuous logging while charging to see if I can find something that reports charging numbers.

WillCodeForCats, 2025-12-24:
Yep, that's the primary reason I created this integration: I have two inverters and no other integration I could find can support more than one device ID over modbus. My integration can do up to 32 device IDs per hub.

Right now if you configure it to `1,3` it can only assume it's an inverter and try to address it with all of the normal inverter locations, but I could make it only do the first 65 sunspec addresses when the model matches known EVSE model strings.

Turn on debug for the integration in HA and then reconfigure it for `1,3` and see what the debug logs show when it tries all the setup functions on the EVSE.

briadelour, 2025-12-25:
> Yep, that's the primary reason I created this integration: I have two inverters and no other integration I could find can support more than one device ID over modbus. My integration can do up to 32 [...]

[home-assistant_solaredge_modbus_multi_2025-12-25T03-41-59.657Z.log](https://github.com/user-attachments/files/24335104/home-assistant_solaredge_modbus_multi_2025-12-25T03-41-59.657Z.log)

briadelour, 2025-12-25:
[config_entry-solaredge_modbus_multi-1b8b6cdae0d78c0084fd51331dbb7934.json](https://github.com/user-attachments/files/24338492/config_entry-solaredge_modbus_multi-1b8b6cdae0d78c0084fd51331dbb7934.json)
[home-assistant_solaredge_modbus_multi_2025-12-25T03-41-59.657Z.log](https://github.com/user-attachments/files/24338491/home-assistant_solaredge_modbus_multi_2025-12-25T03-41-59.657Z.log)

WillCodeForCats, 2026-04-24:
Great, that's a start. There isn't any code to tell the difference between an inverter and EVSE so everything detected is added as an inverter. But it should show the model correctly (SE-EV-SA-US-40P).

Now that it's added as a device can you try getting different diagnostic files from it in different states like idle/disconnected, connected but not charging, and connected charging? I want to see if any of the values it's currently reading for "inverter" change or not.

briadelour, 2026-04-24:
Here are two more

[CHARGING-config_entry-solaredge_modbus_multi-01KQ0D7VDC9YRM8DCW6EEERZ89.json](https://github.com/user-attachments/files/27069300/CHARGING-config_entry-solaredge_modbus_multi-01KQ0D7VDC9YRM8DCW6EEERZ89.json)

[IDLE-PLUGGEDIN-config_entry-solaredge_modbus_multi-01KQ0D7VDC9YRM8DCW6EEERZ89.json](https://github.com/user-attachments/files/27069303/IDLE-PLUGGEDIN-config_entry-solaredge_modbus_multi-01KQ0D7VDC9YRM8DCW6EEERZ89.json)

WillCodeForCats, 2026-05-03:
evse.0 doesn't have the scanning detection for modbus devices, but you're can manually enter the evse id if you know it.

pre.1 and pre.2 has scanning but doesn't have the evse model detection.

A future release will combine both.

nebulight, 2026-05-03:
Got it. I just now installed this: https://github.com/niktest/solaredge_ev_charger_au it appears to be working. Can I grab the evse id from there (or the charger web portal), or would this be a modbus address? I'm just an enduser but willing to help where I can.

WillCodeForCats, 2026-05-03:
It's the modbus address. SolarEdge usually calls it Device ID. It should be You can try ID 3 or 4. If you can access the web interface for the evse then it might be under commissioning -> communication -> rs485 if it's similar to inverters

Or install [Release v3.3.0-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.3.0-pre.2) which has scanning, add the new hub and scan your inverter to see if it finds it (it will show up as an inverter in this version), then install [Release v3.3.0-evse.0](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.3.0-evse.0) last.

nebulight, 2026-05-03:
yep, it was three. However when I installed pre.2 it found it and added it. I then updated to evse.0 and I'm getting this error as now it doesn't see it:

[image]

and confirmed my device ID: 

[image]

nebulight, 2026-05-04:
[image]

Looks to be a similar issues with a slightly different error in the screenshot.

[home-assistant_solaredge_modbus_multi_2026-05-04T01-42-06.582Z.log](https://github.com/user-attachments/files/27327695/home-assistant_solaredge_modbus_multi_2026-05-04T01-42-06.582Z.log)

[14 more replies]