WillCodeForCats/solaredge-modbus-multi #573: Can't find my inverter after install

lbreggi, 2024-04-02:
### Describe the bug

Hi! I'm getting an error after the instalation of this integration. it completes the instalation but can't find my inverter, meters or battery.

### Expected behavior

Be able to add my inverter and the other components of the solar system

### Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Diagnostic File

[home-assistant_solaredge_modbus_multi_2024-04-02T18-41-43.830Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14842455/home-assistant_solaredge_modbus_multi_2024-04-02T18-41-43.830Z.log)

### Home Assistant Version

2024.3.3

### solaredge-modbus-multi Version

2.4.13

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-04-02:
Your inverter stops responding and is timing out the integration:

```
2024-04-02 13:41:26.298 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 40 40
2024-04-02 13:41:38.311 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 12.345 seconds (success: False)
```

The long delay between 13:41:26 and 13:41:38 should be where the integration polls the inverter for data.

Try turning off the option "Auto-Detect Additional Entities" under "Configure" button in the integration panel.

lbreggi, 2024-04-02:
You are the best! now I was able to detect the meters, but can't find the battery. I believe there is a way to extend the time out query? thanks!!

WillCodeForCats, 2024-04-02:
You have battery turned off.

lbreggi, 2024-04-02:
yes, the issue is that I'm getting the time out again when I turn the option to look for the battery on

WillCodeForCats, 2024-04-02:
Unfortunately that's not an integration problem I can fix, if your inverter does not respond to some modbus registers then that might be a firmware issue.

lbreggi, 2024-04-02:
got it.. anyway thanks for your help! just to share I'm running the latest version of the firmware, just in case if someone else face the problem (there is a screenshot with all the versions at the top). thanks again!

WillCodeForCats, 2024-04-02:
You should read this discussion where other people have issues with battery related registers not working on some inverter models: [Battery integration #63](https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/63)

The "Auto-Detect Additional Entities" setting is related to battery registers, so if that doesn't work then it makes sense that batteries won't work either.

WillCodeForCats, 2024-04-02:
Ref: issue #394

WillCodeForCats, 2024-04-02:
Ref: PR #416