WillCodeForCats/solaredge-modbus-multi #574: Battery export/import show as Unknown

billyburly, 2024-04-06:
### Describe the bug

The battery Energy Export and Energy Import are showing up as Unknown. All other values are showing correctly

### Expected behavior

They have valid values

### Screenshots

[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-72d8136f7c782b1ae0807a72090ef8ff.json](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14892100/config_entry-solaredge_modbus_multi-72d8136f7c782b1ae0807a72090ef8ff.json)

### Debug logs

```shell
2024-04-05 22:57:46.585 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 86 86
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_MaxChargePower 0x459c4000 <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_MaxDischargePower 0x459c4000 <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_MaxChargePeakPower 0x459c4000 <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_MaxDischargePeakPower 0x459c4000 <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Temp_Average 0x4177657e <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Temp_Max 0x0 <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_DC_Voltage 0x43c6d2d6 <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_DC_Current 0x3c3e3cbf <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_DC_Power 0xc1a80000 <class 'float'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x10114 <class 'int'>
2024-04-05 22:57:46.586 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Import_Energy_WH 0x10b01 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Energy_Max 0x46179000 <class 'float'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Energy_Available 0x4615199a <class 'float'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_SOH 0x42c80000 <class 'float'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_SOE 0x41989d9e <class 'float'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Status 0x4 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Status_Vendor 0x2 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log1 0x0 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log2 0x0 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log3 0x0 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log4 0x0 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log5 0x0 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log6 0x0 <class 'int'>
2024-04-05 22:57:46.587 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log7 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log8 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor1 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor2 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor3 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor4 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor5 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor6 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor7 0x0 <class 'int'>
2024-04-05 22:57:46.588 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor8 0x0 <class 'int'>
```

### Home Assistant Version

2024.3.1

### solaredge-modbus-multi Version

v2.4.12

### Installation Type

Core

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

### Additional Context

May be related to the March 4.20.32 firmware update. https://www.solaredge.com/us/support/firmware-release-notes/residential-inverters-setapp

I have some test python code that seems to be able to read the battery registers
```
batt 1 lifetime export 0xE176
79938897680793600
batt 1 lifetime import 0xE17A
792915013688885248
```

WillCodeForCats, 2024-04-06:
Why do you think this is a bug and not one of the known issue conditions?

WillCodeForCats, 2024-04-06:
There's only three ways battery energy shows unavailable: if it's zero (not accumulated), if it's max int (`0xFFFFFFFFFFFFFFFF`) or if the value went backwards as described in the docs.

billyburly, 2024-04-06:
> Why do you think this is a bug and not one of the known issue conditions?

Thought it was a bug because I can see the battery. I previously had been effected by the known issue, but moved to a SPAN panel and in the process got battery access enabled.

I had been digging into the for a day but for some reason hadn't restarted HA. Went and updated and restarted and it resolved itself. Looks like something related to updating the inverter/battery firmware got HA and the integration confused, but once restarted it was fine. Wish I had tried a restart earlier

closed 2024-04-06 by billyburly

WillCodeForCats, 2024-04-06:
When you restart HA the "backwards" energy detection is initialized to values read at startup.

This is documented in known issues, not a bug.

billyburly, 2024-04-06:
> This is documented in known issues, not a bug.

Didn't realize that. Spent so much time trying to get solaredge to give me access to my battery I convinced myself thats the only known issues with the batteries. Sorry for the spurious bug report.