WillCodeForCats/solaredge-modbus-multi #578: Battery Power vs. consumption

MikeTango, 2024-04-08:
**Describe the current behavior**
Hi,

first of all thanks for the awesome work. I rarely have any linux stuff that just werks. What i am seeing is that the templated sensors disagree quite a bit on some values, i guess due to the behaviour of the "native" sensors in the system. At night, under battery power, the calculated sensor.power_battery_discharging reads much higher than the calculated sensor.power_consumption. I don't know if this can be attributed to inverter loss.

**What should be updated or changed?**
Possibly making a template sensor that fails over to the higher value at bat discharge. If it is inverter loss, this is moot.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: newest
 - solaredge-modbus-multi Version: newest

**Additional context**
Add any other context about the request here.

MikeTango, 2024-04-08:
Hi,

i am very new to this and the solar system isn't even two days old, so there's not too much data. Here's last night. The rhythmic highs (and the crazy transients) are probably from the fridge(s) or the hot water circulation pump. Battery is always ~100W above the consumption.

[image]
[image]

MikeTango, 2024-04-08:
That sounds like you have a stack of three batteries but are only using the data of one.

WillCodeForCats, 2024-04-08:
This isn't an integration issue so I'm going to move it to discussions.

Templates in the wiki are contributed (not by me): please contribute new templates as appropriate. I don't have batteries personally so I can't help.