WillCodeForCats/solaredge-modbus-multi #581: Battery export sensor incorrect

jongbj1972, 2024-04-09:
### Describe the bug

The battery export sensor indicates a battery discharge of only 3kWh/day while reality is approx. 12kWh/day

### Expected behavior

correct data being read from the battery export sensor

### Screenshots

[image]
[image]
[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-a150630511e990a73146f8eb62c16c53 (1).json](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/14922422/config_entry-solaredge_modbus_multi-a150630511e990a73146f8eb62c16c53.1.json)

### Debug logs

```shell
20 minute debug log was almost 4 million characters and messed up the form.
```

### Home Assistant Version

2024.4.1

### solaredge-modbus-multi Version

latest

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

### Additional Context

Up until recently, the battery export sensor functioned fine, although, due to a database change, I cannot dettermine when it started to go wrong, but it is clearly visible that the data is not correct.

WillCodeForCats, 2024-04-09:
If you look in the debug file look for the `battery_id_1` section towards the bottom.

The sensor values for battery energy are:

```
        "B_Export_Energy_WH": "0x40806d",
        "B_Import_Energy_WH": "0x409e8b",
```

These are hex integer values, so a value of `0x40806d` is 4227181 watt-hours, or 4227.181 kilowatt-hours

The integration does not interpret these values, they are passed through as given by the inverter. You would need to either download diagnostic files or watch debug logs to check if that value is increasing or not.

The data in the SolarEdge online platform is separate from modbus.

jongbj1972, 2024-04-10:
So if I understand you correctly, the integration reads the values as presented by the inverter and only converts the HEX to numerical value. Then I appologize for sending in the bug report and will try to solve on the inverter side.

deqw, 2024-04-19:
Concerning these two sensors (sensor.solaredge_b1_energy_import and sensor.solaredge_b1_energy_export) I see them unavailable frequently. Reloading the integration kicks them again.
[image]

richardambridge, 2024-04-20:
Did you find a solution for this?
I see the same

[image]

I'm running off battery overnight, and the B1 Energy export level is not changing overnight

jongbj1972, 2024-04-20:
No, according to the developer it is not the integration, so no, there is no solution. 
I doubt it is the SolarEdge software as the web-application shows the correct information.

WillCodeForCats, 2024-04-20:
Download diagnostic files or check debug logs to see if the value of `B_Export_Energy_WH` or `B_Import_Energy_WH` is changing.

MikeTango, 2024-04-20:
Hi,

i am having the same issue and will monitor the logfiles after it freezes, then reload and check the logfiles again.

richardambridge, 2024-04-21:
With debug on, the values are not changing overnight,  but I see the data in the solaredge app is changing.

`2024-04-21 04:00:47.479 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:01:02.329 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:01:17.428 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:01:32.341 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:01:47.365 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:02:02.363 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:02:17.402 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:02:32.371 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:02:47.339 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:03:02.491 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:03:17.446 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:03:32.369 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:03:47.416 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:04:02.290 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:04:17.387 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:04:32.411 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:04:47.391 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:05:02.461 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:05:17.399 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:05:32.420 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:05:47.374 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:06:02.391 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:06:17.387 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:06:32.360 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:06:47.388 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:07:02.390 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:07:17.406 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:07:32.359 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:07:47.405 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:08:02.512 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:08:17.315 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:08:32.362 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:08:47.376 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:09:02.359 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:09:17.411 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:09:32.528 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:09:47.356 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:10:02.540 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:10:17.367 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
[... cut ...] 32.340 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:10:47.418 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>
2024-04-21 04:11:02.361 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0x14eaf0 <class 'int'>`

[image]

although their website is showing 0W for the time, but it does show some red data ?

[image]

richardambridge, 2024-04-21:
[04.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/15051286/04.log)

here is all the data for 04:xx hour:   grep custom_components.solaredge_modbus_multi.hub home-assistant.log |grep " 04:..:.." > /tmp/04.log

MikeTango, 2024-04-21:
Step 1: Screenshot, state unknown since yesterday 2200ish
Step 2: Activate logging
Step 3: Reset Integration
Step 4: Screenshot, state OK
Step 5: Download log
Step 6: Deactivate logging

The first lines in the log :

`2024-04-20 22:24:14.794 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2024-04-20 22:24:18.599 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2024-04-20 22:24:24.002 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2024-04-20 22:24:34.205 WARNING (MainThread) [pymodbus.logging] Failed to connect ` 

Correspond well with the battery sensor death, although i don't know whether they are zulu time or local (+2). I might have screwed up the logging/resetting process since the log shows nothing about the integration restarting. 

[home-assistant_solaredge_modbus_multi_2024-04-21T09-35-53.544Z.log](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/15051326/home-assistant_solaredge_modbus_multi_2024-04-21T09-35-53.544Z.log)

`024-04-21 11:33:46.248 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Export_Energy_WH 0xa6b <class 'int'>
2024-04-21 11:33:46.248 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Import_Energy_WH 0xe72 <class 'int'>`

WillCodeForCats, 2024-04-21:
So first off, all I'm really doing here is proving this is not a bug in the integration: the integration just takes values the inverter gives, does some validation and checks for obviously wrong or invalid states that would mess up graphs/statistics, and passes them into Home Assistant.

The SolarEdge app and monitoring platform is completely separate from modbus. If one of those works and the other doesn't, that's a bug in the inverter firmware that SolarEdge would need to fix. This integration can only use the data the inverter gives it over modbus. If the inverter keeps returning the same data every time it asks without increasing the values, that's an inverter bug and unfortunately there's nothing I can do about it except show you how it's not my problem to fix.

You could try power cycling your entire system to cold boot it (turn off all AC and DC safety switches) to see if that re-syncs everything. That's been reported as a solution in the past and is mentioned in known issues.

Battery energy sensors going unavailable can be several things which are already documented in known issues:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues

If you have battery energy going unavailable you probably have the "backwards" data issue and need to turn on some of the options that are designed to allow that. Home Assistant expects that energy meters should not go backwards, so if they do the sensors will go into unavailable state.

MikeTango, 2024-04-21:
hi,i am running an original SolarEdge battery so it seems to be either something different or they have this problem with their own batteries now.

WillCodeForCats, 2024-04-21:
> Correspond well with the battery sensor death, although i don't know whether they are zulu time or local (+2). I might have screwed up the logging/resetting process since the log shows nothing about the integration restarting. It also starts timestamping after my screenshot. Do i need to download the log before restarting?

Yes because logs are cleared when Home Assistant restarts.

MikeTango, 2024-04-21:
I restarted the integration only, there was no new boot info in the log.

WillCodeForCats, 2024-04-21:
I'm not really seeing anything here that suggests a problem with the integration.

I don't have batteries on my system, so I can't give any advice on them.