WillCodeForCats/solaredge-modbus-multi #585: SE10K-RWB48BFN4 - ID1 wrong AC Power

gitmawa, 2024-04-18:
### Describe the bug

I have 2 inverters in use, SE10K-RWB48BFN4 with ID1 and SE10K-RW0TEBEN4 with ID 2
On the first inverter SE10K-RWB48BFN4 Hybrid I see "AC Power **-127,220 W**"
and on the second inverter SE10K-RW0TEBEN4 "AC Power**1,267,400 W**"
My mobile app shows 2,800 W

### Expected behavior

-127.220 W makes no sense

### Screenshots

[image]
[image]

### Diagnostic File

i don't have

### Home Assistant Version

2024.4.2

### solaredge-modbus-multi Version

2.4.13

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-04-18:
You have to upload your diagnostic file.

suelm, 2024-04-18:
I have the same problem.
One Master RWB and one Follower RWB.
These use RS485-2 for the follower and not RS485-1.
If I specify in the configuration that I have two inverters, the setup fails. If I specify only one inverter, it is found, but with the wrong values

WillCodeForCats, 2024-04-18:
> I have the same problem. One Master RWB and one Follower RWB. These use RS485-2 for the follower and not RS485-1. If I specify in the configuration that I have two inverters, the setup fails. If I specify only one inverter, it is found, but with the wrong values

This is not the same problem. Read the setup instructions first.

WillCodeForCats, 2024-04-18:
I'm going to close out this issue becasue it is incomplete, please resubmit with your diagnostic file because that's where the answer is.

If I ask for something that means it's not optional. The only time you would not provide a diagnostic file is if the integration won't start.

closed 2024-04-18 by WillCodeForCats

suelm, 2024-04-18:
> This is not the same problem. Read the setup instructions first.

Ok, then it's a different problem.
I have read the instructions x times. I can't find an error and can't get a connection.

WillCodeForCats, 2024-04-18:
In order to keep issue reports on topic, I ask that any "help" requests must go into discussions or the Home Assistant forums. I can't answer setup help questions in issues.

WillCodeForCats, 2024-04-18:
As far as this issue goes: it is possible for inverters to report negative numbers.

Negative inverter power AC/DC values are valid in certain battery charging conditions.

Export+Import meters will return positive power values during export and negative values during import.

Modbus only reports data from the inverters, it doesn't give interpreted values like the Solaredge monitoring platform does. There are templates for this, and probably you can find them in the discussions. I don't personally have batteries.

gitmawa, 2024-04-19:
I have found the solution for myself.
The power production (AC power) from the **hybrid inverter** delivers different values than expected when the battery is charged, as it is also connected to this inverter. This can even go so far as to give negative kW values. I now simply add the current flow into the battery and the value from the hybrid inverter together and my values are correct.

```
- sensor:
    - name: "SolarEdge - Solarstrom Hybrid"
      unique_id: "Solarstrom_Hybrid"
      unit_of_measurement: "W"
      state_class: total_increasing
      device_class: "power"
      state: '{{ (states("sensor.solaredge_i1_ac_power") | round | int + states("sensor.solaredge_b1_dc_power") | round | int) }}'
```

gitmawa, 2024-04-19:
This is the code for calculating the current power generation of the panels with both inverters

`- sensor:
    - name: "SolarEdge - Solarstrom i1 i2 b1"
      unique_id: "Solarstrom_i1_i2_b1"
      unit_of_measurement: "W"
      state_class: total_increasing
      device_class: "power"
      state: '{{ (states("sensor.solaredge_i1_ac_power") | round | int + states("sensor.solaredge_i2_ac_power") | round | int + states("sensor.solaredge_b1_dc_power") | round | int) }}'`