WillCodeForCats/solaredge-modbus-multi #595: After update to latest version Error: "list indices must be integers or slices, not str"

gapx15, 2024-05-02:
### Describe the bug

After updating to the latest version there is an error: "list indices must be integers or slices, not str" I tried to reinstall it but no luck.

### Expected behavior

Before update, there was no error.

### Screenshots

[image]

### Diagnostic File

list indices must be integers or slices, not str

### Home Assistant Version

Core: 2024.5.0 Frontend: 20240501.0

### solaredge-modbus-multi Version

latest

### Installation Type

Container

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-05-02:
Please try [Release v2.4.15-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.15-pre.1)

Upload your diagnostic file if this pre-release works.

WillCodeForCats, 2024-05-03:
The full release for [Release v2.4.15](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.15) is now available.

With your Synergy inverter this release will add new devices that show the status of the modules.