WillCodeForCats/solaredge-modbus-multi #601: Battery only has "Watt" Sensors and not "Watt-Hours" Sensors

tarkin2020, 2024-05-20:
### Describe the bug

As said, the battery sensors has no Watt-Hours Sensor. There is only the current Power showd (It is in watt) thierfor some can not use this sensor in the HA Dashboard

### Expected behavior

Two Sensors:
Energey Exported unit: kWh
Energy Imported unit: kWh

### Screenshots

[image]

### Diagnostic File

[image]

### Home Assistant Version

2024.5.4

### solaredge-modbus-multi Version

2.4.15

### Installation Type

HAOS

### Read the Instructions

- [X] I have read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki
- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] This issue is not about a template error (template errors should not be reported as an issue, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-05-20:
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues

elad-bar, 2024-05-22:
@WillCodeForCats can you pls refer to the specific known issue?
i have the same issue and would like to understand how to address it

thanks

WillCodeForCats, 2024-05-22:
And these options:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration#battery-options

elad-bar, 2024-05-22:
I understand, my issue is not related to the battery but to general measurements that are w and not wh,
For instance Inverter DC Power,
Reason i'm asking is for Home Assistant energy module that current solar production is not available in wh just w

WillCodeForCats, 2024-05-22:
DC Power sensor is watts because power is measured in watts.

https://en.wikipedia.org/wiki/Electric_power

Why do you want to use a Power sensor in the energy dashboard? That's just not possible. It's called the "energy" dashboard because you need to use a sensor that provides Energy values.