WillCodeForCats/solaredge-modbus-multi #606: Inconsistent battery energy data

IngmarStein, 2024-06-04:
### Describe the bug

I have two inverters, a SE10K-RWB (hybrid) and a SE10K (not hybrid). A SolarEdge battery is connected to the hybrid inverter. The energy meter is a SE-MTR-3Y-400V-A.

I have read that some battery energy meters regularly reset to 0 and others are ever-increasing.

In my installation, I get sparse readings which randomly increase and decrease:

The other energy meters look fine (grid import/export are continuously increasing lines).

### Expected behavior

I expect the energy meters to increase over time so that I can extract a daily min/max and compute the charged/discharged energy on a daily basis.

### Screenshots

[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-7b87ebcb4b1fe3c518722fe6c09782cf.json](https://github.com/user-attachments/files/15551339/config_entry-solaredge_modbus_multi-7b87ebcb4b1fe3c518722fe6c09782cf.json)

### Home Assistant Version

2024.5.5

### solaredge-modbus-multi Version

2.4.15

### Installation Type

Container

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-06-04:
This is already documented in Known Issues. Only SolarEdge can fix battery energy.

closed 2024-06-04 by WillCodeForCats

WillCodeForCats, 2024-06-04:
I remember seeing someone use an Integration helper (Riemann sum integral sensor) to approximate battery energy from unreliable data. You could try that. I don't have a link though since I don't remember if I saw it in github discussions or on the HA forums.

IngmarStein, 2024-06-04:
In that case, I apologize for asking about a well-known issue. If you don't mind, I'll add the SolarEdge Home Battery 48V to the list of known problematic products (it's separate from the high-voltage Energy Bank).

I've createdd a support ticket but don't hold my breath. In the meantime, computing a Riemann sum integral from the DC power measurement seems like a suitable workaround (compromising precision, of course).