WillCodeForCats/solaredge-modbus-multi #610: Battery sensors 'unknown'

rploeg, 2024-06-26:
### Describe the bug

the battery sensor import and export that I use for my energy dashboard don't work anymore:

### Expected behavior

import in battery and export in battery

### Screenshots

[image]

### Diagnostic File

let me know what you need. 

### Home Assistant Version

2024.6.4

### solaredge-modbus-multi Version

2.4.15

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-06-26:
I'm sure this is the same reason it always is:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues#battery-energy

Next release is getting a logging message. I can't do anything about battery energy meters beyond the settings the integration already has unless someone has a new idea on how to handle it.

WillCodeForCats, 2024-06-26:
[Release v2.4.16-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.16-pre.2) adds logging for the energy values go backwards condition.

rploeg, 2024-06-26:
Thanks! I will test that one.

closed 2024-06-26 by rploeg

WillCodeForCats, 2024-06-26:
It's just going to log, there is no fix I'm aware of unless SolarEdge fixes their firmware to make battery energy a true increasing value.