WillCodeForCats/solaredge-modbus-multi #624: new BinarySensor "Grid Status" should be BinarySensorDeviceClass = Energy

ChristophCaina, 2024-07-14:
**Describe the current behavior**
The current BinarySensor "Grid Status" does not have a default BinarySensorDeviceClass.
You can select the deviceClass in the Entity Settings, but I think, it should be "Energy" by default.
By having a specific DeviceClass assigned, the BinarySensor can be selected as alert-class in the Area-Card, like shown in this screenshot.

[image]

This Icon will be shown, whenever the Sensor is "ON"

Since an alarm should be displayed whenever there's a "problem", I think the logic should be inverted - but this would be something that needs to be done from HomeAssistant.

**What should be updated or changed?**
The Binary Sensor "Grid Status" should have a default BinarySensorDeviceClass, preferred "Energy"

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2024.7.2
 - solaredge-modbus-multi Version: 2.4.17

**Additional context**
Add any other context about the request here.

WillCodeForCats, 2024-07-14:
ENERGY is not a supported class for binary sensor.

https://developers.home-assistant.io/docs/core/entity/binary-sensor

POWER is an option though.

ChristophCaina, 2024-07-14:
Sorry, this was a translation mistake, since the deviceClass was translated as "Energy" into german ... :D

WillCodeForCats, 2024-07-14:
Testing the PR and I will merge it as soon as I check it on my test system.

ChristophCaina, 2024-07-14:
thanks :)

I have implemented the changes, and everything seems to work.
Yet, as I have already set the deviceclass, I can't really check the difference ^^

But at least, nothing unexpected observed so far.

closed 2024-07-14 by WillCodeForCats

WillCodeForCats, 2024-07-14:
[Release v2.4.18](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.18)