WillCodeForCats/solaredge-modbus-multi #631: Inverter data no longer reporting via plugin

turtle2472, 2024-07-22:
### Describe the bug

HAOS rPi Install: 2024.7.3 Core, 2024.06.2 Supervisor, 12.4 OS
Inverter FW Version: 4.20.36
Plugin Version: 2.4.18
Configuration: Two separate inverters each mapped via its own IP. (.70 and .71), no RS485 in use.

Error began appearing last night around 2340 EDT in HA and any data from the plugin now reports "Unavailable". Now in HA Settings SolarEdge Multi: "Failed setup, will retry: Modbus error: Modbus Error: [Input/Output] ERROR: No response received after 3 retries"

Connection via telnet to 1502 succeed for both inverters still.

### Expected behavior

All data was reporting normally until last night at 2340 EDT.  Logs show heat sink temps, AC/DC production values all normally.  All reporting stopped and became unavailable after that.  Here is a screenshot demonstrating the good and the last recorded value.

### Screenshots

[image]
[image]

### Diagnostic File

I've attempted to follow your guide for gathering the diagnostic file but there is not option for the three dots with "Download Diagnostics".

### Debug logs

```shell
[core-ssh homeassistant]$ cat home-assistant.log|grep solaredge_modbus_multi
2024-07-22 17:03:07.821 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solaredge_modbus_multi which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-07-22 17:50:04.847 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=False, detect_extras=True, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
2024-07-22 17:50:04.848 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Retry limit=5 time=800 ratio=3
2024-07-22 17:50:04.848 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] New client object for 192.168.0.70:1502
2024-07-22 17:50:04.848 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] reconnect_delay=0 reconnect_delay_max=3 retry_on_empty=False timeout=3
2024-07-22 17:50:04.859 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2024-07-22 17:50:04.859 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for inverter at 192.168.0.70 ID 1
2024-07-22 17:50:04.888 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 69 69
2024-07-22 17:50:04.889 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_ID 0x53756e53<class 'int'>
2024-07-22 17:50:04.889 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_DID 0x1<class 'int'>
2024-07-22 17:50:04.889 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_Length 0x41<class 'int'>
2024-07-22 17:50:04.890 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_Manufacturer SolarEdge<class 'str'>
2024-07-22 17:50:04.890 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_Model SE7600H-US000BNU4<class 'str'>
2024-07-22 17:50:04.890 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_Option <class 'str'>
2024-07-22 17:50:04.890 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_Version 0004.0020.0036<class 'str'>
2024-07-22 17:50:04.890 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SerialNumber 74xxxxFD<class 'str'>
2024-07-22 17:50:04.891 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_Device_address 0x1<class 'int'>
2024-07-22 17:50:04.919 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 9 9
2024-07-22 17:50:04.920 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1 MMPPT: mmppt_DID 0x2bd <class 'int'>
2024-07-22 17:50:04.920 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1 MMPPT: mmppt_Length 0x99 <class 'int'>
2024-07-22 17:50:04.920 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1 MMPPT: mmppt_Units 0xffff <class 'int'>
2024-07-22 17:50:04.920 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1 is NOT Multiple MPPT
2024-07-22 17:50:04.921 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for meter I1M1
2024-07-22 17:50:04.958 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 67 67
2024-07-22 17:50:04.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_SunSpec_DID 0x2bd <class 'int'>
2024-07-22 17:50:04.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_SunSpec_Length 0x99 <class 'int'>
2024-07-22 17:50:04.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_Manufacturer KLd	 <class 'str'>
2024-07-22 17:50:04.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_Model Z <class 'str'>
2024-07-22 17:50:04.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_Option  <class 'str'>
2024-07-22 17:50:04.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_Version  <class 'str'>
2024-07-22 17:50:04.959 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_SerialNumber  <class 'str'>
2024-07-22 17:50:04.960 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_Device_address 0x8000 <class 'int'>
2024-07-22 17:50:04.960 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: Meter 1 ident incorrect or not installed.
2024-07-22 17:50:04.960 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for meter I1M2
2024-07-22 17:50:04.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 67 67
2024-07-22 17:50:04.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M2: C_SunSpec_DID 0xffff <class 'int'>
2024-07-22 17:50:04.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M2: C_SunSpec_Length 0x0 <class 'int'>
2024-07-22 17:50:04.979 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M2: C_Manufacturer  <class 'str'>
2024-07-22 17:50:04.979 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M2: C_Model  <class 'str'>
2024-07-22 17:50:04.979 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M2: C_Option  <class 'str'>
2024-07-22 17:50:04.979 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M2: C_Version  <class 'str'>
2024-07-22 17:50:04.979 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M2: C_SerialNumber A <class 'str'>
[... 71 lines cut ...]
2024-07-22 17:58:15.642 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 40 40
2024-07-22 17:58:27.652 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 12.245 seconds (success: False)
```

### Home Assistant Version

2024.7.3

### solaredge-modbus-multi Version

2.4.18

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

### Additional Context

Again, I've been using this plugin for more than 6 months and all was good until last night.  I do not have anything else attempting to access modbusTCP on the inverters, just this plugin.

WillCodeForCats, 2024-07-22:
Turn off auto detect additional entities.

turtle2472, 2024-07-22:
That did it.  Thank you!

closed 2024-07-22 by turtle2472

WillCodeForCats, 2024-07-22:
It's an inverter firmware issue. SoarEdge changed the modbus behavior for unsupported features. If it worked before you probably got a firmware update.

turtle2472, 2024-07-23:
Yeah, my inverters were upgraded last week.