WillCodeForCats/solaredge-modbus-multi #637: Automatic inverter ID detection for setup

WillCodeForCats, 2024-07-27:
Add an option to auto detect inverter modbus device IDs or configure manually when setting up the integration.

WillCodeForCats, 2024-07-27:
Possible issue: If EV chargers also show up as device IDs, there does not appears to be a way to determine if the detected ID is an inverter or a charger. (Issue #565)

ansgarkewitz, 2025-04-06:
Hello  I have two inverters one master one slave. Both have a batterie. On the master the meter and the battery are connected to the first rs 485-1 . The second inverter is connected to the second rs 485-2 with the  id 2. The second inverter is connected with the first rs 485-1 to the batterie and with the second rs485-2 to the first having id 2. The first ineverter has id 1. This config does not follow the manual of the integration but is working with solaredge. I have deleted the hub and installed it again. No difference.