WillCodeForCats/solaredge-modbus-multi #638: Two separate SE Inverters not possible over ModBusProxy

Stony111, 2024-07-30:
### Describe the bug

System architecture is like follow descirbed:
- SE Inverter 1 with Meter and Battery on IP-Adress 192.168.0.1 - Port 1502 (for example)
- SE Inverter 2 (no Meter no Battery) on IP-Adress 192.168.0.2 - Port 1502 (for example)
- Both Inverters NOT connected together via RS485 - both runs complete separate
- ModbusProxy, which manage both IP-Adresses; IP-Adress of ModBusProxy Instance 192.168.0.3 (for example)
++ Inverter 1 is listen on port 2502 on this proxy
++ Inverter 2 is listen on port 2503 on this proxy
- Home Assistant with this integration

When I now want to configure the second inverter, the IP-Adress is the same, as of Inverter 1, because it's always the IP-Adress of ModBusProxy-Instance 192.168.0.3 - but the Port would be different: 2502 for Inverter 1 and 2503 for Inverter 2.

When I want to save the new Hub Configuration an error appears: Inverter is already configured!
I think in this check only IP-Adress is validaten not the combination of IP-Adress and Port

### Expected behavior

Creating a complete new Hub Instance of a second standalone Inverter, when Connection to all Inverters are handled via ModBusProxy!

### Screenshots

[image]

### Diagnostic File

There is no diagnostic file - its only a UI / check Problem.

### Home Assistant Version

2024.7.3

### solaredge-modbus-multi Version

2.4.18

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-07-30:
The integration currently uses the host or IP to determine duplicates, even if the port is different.

It's on the to-do list to update this

Stony111, 2024-07-31:
Thanks a lot for the fast info. Do you have a plan for when you would implement it? Just so I can judge whether it's worth waiting ;-)

WillCodeForCats, 2024-08-08:
I don't have an estimated time right now, but I would probably work on it after the non-sequential inverter ID changes.

WillCodeForCats, 2025-01-19:
Pull request #737 will migrate the Home Assistant internal config to use the address string (host:port) as the unique ID for a device hub.

WillCodeForCats, 2025-01-19:
[Release v3.0.6-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.0.6-pre.1)

closed 2025-01-27 by WillCodeForCats