WillCodeForCats/solaredge-modbus-multi #641: Unexpected Keyword "retry_on_empty" while init AsyncModbusTcpClient

Stony111, 2024-08-02:
### Describe the bug

After installing an removing a other HACS integration for reading solaredge inverters a error appears while starting your solaredge-modbus-multi integration on HA startup:

[image]

I think the problem is, that with this other integration an newer Version of pymodbus-lib came in my system :-(((

So I have searched with google for a possible cause an find a documentation for the latest pymodbus API. There is described that the parameter retry_on_empty was removed (I hope thats the correct finding ;-))

[image]

This are my HA-Versions:
[image]

But your really super extension worked with this HA-Version until I installed an other HACS-solaredge-integration :-(((

What is your opinion on the subject? Have you a tipp for my how I can determine the installed version of pymodus-lib in my system?

Thanks a lot!

### Expected behavior

No Error while startup process of integration!

### Diagnostic File

No File available

### Debug logs

```shell
Logger: custom_components.solaredge_modbus_multi
Quelle: helpers/update_coordinator.py:312
Integration: SolarEdge Modbus Multi (Dokumentation, Probleme)
Erstmals aufgetreten: 08:04:25 (17 Vorkommnisse)
Zuletzt protokolliert: 08:21:53

Unexpected error fetching SolarEdge Coordinator data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 190, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 234, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 231, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 364, in async_refresh_modbus_data
    await self.connect()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 470, in connect
    self._client = AsyncModbusTcpClient(
                   ^^^^^^^^^^^^^^^^^^^^^
TypeError: AsyncModbusTcpClient.__init__() got an unexpected keyword argument 'retry_on_empty'
```

### Home Assistant Version

2024.7.4

### solaredge-modbus-multi Version

2.4.18

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

dirixmjm, 2024-08-02:
I'm having the same issue with pymodbus 3.7.0 and HA 2024.7.4. The integration worked for a while (although changing power settings I have issues understanding how it works, sometimes I could get it to work and sometimes it just did not respond).
Anyway, right now the integration is not starting / loading at all with the same error as listed here.

Stony111, 2024-08-02:
By-the-way: my dirty-hack to resolve the error in my system landscape is like this:

Commet out the line 475 in hub.py:
[image]

After restart HA integration works fine again - but this can only be a temporarily solution and I don't know what other effects the missing option has. But at first glance, everything seems to be running normally for me!

WillCodeForCats, 2024-08-02:
Yes, installing something external that forces the pymodbus version to 3.7.0 will cause it to break since that parameter was removed. Home Assistant core uses `pymodbus==3.6.9`. I will remove it from the modbus client.

closed 2024-08-02 by WillCodeForCats