WillCodeForCats/solaredge-modbus-multi #655: Make step more precise

HexiDev, 2024-08-20:
**Describe the Current Behavior:**
The Active Power Limit slider currently adjusts in increments of 1.

**Suggested Change:**
The current step size of 1 is too coarse, making it difficult to precisely adjust the output in watts. This lack of precision can result in either too much or too little power, preventing near-perfect control and causing grid leakage.

**Home Assistant Details:**
- Home Assistant Core Version: Latest
- solaredge-modbus-multi Version: Latest

WillCodeForCats, 2024-08-20:
Active power limit is an unsigned integer with a range of 0-100% That's all you get out of that setting.

WillCodeForCats, 2024-08-20:
A better way to limit export is using with site limit settings, which you can set a value in watts. But this isn't something I do personally, so you might want to check in discussions for help.

As a feature request I can't do this because the inverter does not support anything other than steps of 1 on this this setting.

closed 2024-08-20 by WillCodeForCats

HexiDev, 2024-08-20:
Ah what a bummer, sadly in newer version's the have locked down the site access and will only be able to connect in AP mode. (Thanks for the fast response btw)