WillCodeForCats/solaredge-modbus-multi #662: Use UnitOfReactivePower class enum

WillCodeForCats, 2024-08-28:
Use UnitOfReactivePower class enum introduced in HA 2024.9.0

skipperkm, 2024-09-25:
Logger: homeassistant.const
Quelle: helpers/deprecation.py:197
Erstmals aufgetreten: 15:24:49 (1 Vorkommnisse)
Zuletzt protokolliert: 15:24:49

POWER_VOLT_AMPERE_REACTIVE was used from solaredge_modbus_multi, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'solaredge_modbus_multi' custom integration

WillCodeForCats, 2024-09-25:
I'm not planning to merge the existing PR until HA 2024.10 or 2024.11 because I don't want to force a minimum version of HA for everyone who doesn't want to be on the latest HA just to get a bug fix in the integration.

closed 2024-10-14 by WillCodeForCats