WillCodeForCats/solaredge-modbus-multi #669: POWER_VOLT_AMPERE_REACTIVE

nagtegaal, 2024-09-08:
### Describe the bug

In the logs it states:
`
Logger: homeassistant.const
Bron: helpers/deprecation.py:197
Eerst voorgekomen: 09:32:07 (1 gebeurtenissen)
Laatst gelogd: 09:32:07
POWER_VOLT_AMPERE_REACTIVE was used from solaredge_modbus_multi, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'solaredge_modbus_multi' custom integration`

### Expected behavior

No error or warning messages in the logs about deprecated functions.

### Diagnostic File

Not needed. It's an HA api change which should be respected and followed. By preference, before hit starts to complain about the deprecation of certain functions and services

### Home Assistant Version

2024.9

### solaredge-modbus-multi Version

2.4.20

### Installation Type

Supervised

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-09-08:
Duplicate.

WillCodeForCats, 2024-09-25:
Duplicate of #662

closed 2024-09-25 by WillCodeForCats