WillCodeForCats/solaredge-modbus-multi #680: depricated call

ChristophCaina, 2024-09-25:
### Describe the bug

the following warning will be logged:

```
Logger: homeassistant.const
Quelle: helpers/deprecation.py:197
Erstmals aufgetreten: 24. September 2024 um 13:40:04 (1 Vorkommnisse)
Zuletzt protokolliert: 24. September 2024 um 13:40:04

POWER_VOLT_AMPERE_REACTIVE was used from solaredge_modbus_multi, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'solaredge_modbus_multi' custom integration
```

### Expected behavior

integration should be fixed and use the correct constant :)

### Diagnostic File

--

### Home Assistant Version

2024.9.3

### solaredge-modbus-multi Version

3.0.0

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-09-25:
I'm not planning to merge the existing PR until HA 2024.10 or 2024.11 because I don't want to force a minimum version of HA for everyone who doesn't want to be on the latest HA just to get a bug fix in the integration.

ChristophCaina, 2024-09-25:
Hi, Sure, that's fine.
I will add the Changes locally - but would recommend to keep the issue Open until this has been fixed (also for Others) or Not?

WillCodeForCats, 2024-09-25:
There's already an open issue.

WillCodeForCats, 2024-09-25:
Duplicate of #662