WillCodeForCats/solaredge-modbus-multi #687: Modbus freezes from time to time

djfanatix, 2024-10-12:
### Describe the bug

As discussed on the HA community forum, hereby some logs when Solaredge modbus crashes.
I have an inverter with display (upgraded with Setapp, latest FW), a meter and Storedge.

I use a modbus proxy, but the crashes also when I tested the integration with a direct connection without other modbus interface.
I know have an automation to restart the modbus proxy when the freezing occurs.

### Expected behavior

work fine

### Diagnostic File

[solaredgelog.log](https://github.com/user-attachments/files/17349116/solaredgelog.log)

### Debug logs

```shell
see attached, please ignore the template errors. (not relevant)
```

### Home Assistant Version

2024.10.2

### solaredge-modbus-multi Version

3.0.0

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-10-13:
Turn off the "Keep Modbus Open" option.

WillCodeForCats, 2024-10-13:
I also need you to upload your diagnostic file and turn on `pymodbus` debugging:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Integration-Diagnostics

WillCodeForCats, 2024-10-13:
The issue appears to be that `pymodbus` isn't connected after sending a connect call:

```
2024-10-12 07:02:40.521 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Connecting to 192.168.1.120:1502 ...
2024-10-12 07:02:40.527 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 10.8
2024-10-12 07:02:46.532 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-10-12 07:02:46.532 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Disconnectng from 192.168.1.120:1502 (clear_client=False).
```

djfanatix, 2024-10-14:
[home-assistant_solaredge_modbus_multi_2024-10-13T19-34-18.165Z.log](https://github.com/user-attachments/files/17360173/home-assistant_solaredge_modbus_multi_2024-10-13T19-34-18.165Z.log)

Hereby with Pymodbus logs attached.

- When it crashes I have an automation to restart the modbus, this usually fixes it again.
- Could it be that some automations are writing too many battery commands? or is this not relevant? (it used to work fine before)
- To be clear, your integration worked fine for a very long time with my very old ancient firmware, it only happened after the new SE firmware and Setapp instead of display control.

drumtrich, 2024-11-05:
> The issue appears to be that `pymodbus` isn't connected after sending a connect call:
>
> ```
> 2024-10-12 07:02:40.521 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Connecting to [...]

I think I have the same problem. Appeard first time a few days ago and today again.

```
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: pymodbus.logging
Quelle: custom_components/solaredge_modbus_multi/hub.py:506
Integration: SolarEdge Modbus Multi (Dokumentation, Probleme)
Erstmals aufgetreten: 11:41:15 (152 Vorkommnisse)
Zuletzt protokolliert: 15:40:11

Cancel send, because not connected!
```

Here my diagnostic file, maybe it can help you to find the problem

[config_entry-solaredge_modbus_multi-01JBEQY3F3CQFE50B72DWABGRF.json](https://github.com/user-attachments/files/17634381/config_entry-solaredge_modbus_multi-01JBEQY3F3CQFE50B72DWABGRF.json)

GpdVatter, 2024-11-24:
The update of the Inline Meter to 1.0.90 fix the bug within my homeassistsnt environment

drumtrich, 2024-11-25:
How to update the inline Meter? My has Firmware: 1.0.81

GpdVatter, 2024-11-25:
> […](#)

with the SolarEdge Inverter SetApp application on mobile device

WillCodeForCats, 2025-04-17:
I've not yet seen modbus freezing personally.

My setup is connected with wired Ethernet and a 30 second polling rate. One meter (hardwired on the inverter 485 bus), no batteries, no commands, no proxies.