WillCodeForCats/solaredge-modbus-multi #691: Inverter charge limit doesn't work at night

FranciscoBigWings, 2024-10-29:
### Describe the bug

When it is cloudy or in winter in genera, I use cheap grid tariff to charge the battery and for that I use the Storage Default Mode to Charge from Solar Power and Grid and used the Charge and Discharge limits to play with the values.

Last winter everything used to work well with this integration (I can't remember the version). Now I have latest version and also in the latest version before 3.0.0. it used to happen that I can modify at night the Discharge Limit but I can't get the Charge Limit to be changed. During the day there are no issues and I in fact change it at sunset to 0W but when at 1AM I want to modify to 1000W for example, it is not possible.

As everything works during the day there could be something I am missing but I couldn't find anything that explains why at night I can't modify this numeric value.

Thanks

### Expected behavior

Modify Charge Limit numeric value at any time of the day

### Diagnostic File

[config_entry-solaredge_modbus_multi-aa74d1e75f85eaa741ca4b8728e80c78.json](https://github.com/user-attachments/files/17564239/config_entry-solaredge_modbus_multi-aa74d1e75f85eaa741ca4b8728e80c78.json)

### Home Assistant Version

24.10.4

### solaredge-modbus-multi Version

3.0.1.

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-10-30:
AC Charge Limit is always available when charge policy is Fixed Energy Limit or Percent of Production. The diagnostic file you uploaded shows a policy of Always Allowed which disables AC Charge limit.

If you are able to set the value then the integration is fine and perhaps someone who has batteries can help answer questions about using setting. I don't have batteries.

FranciscoBigWings, 2024-10-30:
I am sure the integration works well, during the day its perfect.

I tried to change AC Charge Limit from Always Allowed to the other options, even if I dont understand how it would affect the Charge Limit of the battery; the result is the same I can't modify the Charge Limit, the value won't save.

Lets further explain my installation:
- Leader SE inverter is an EV inverter
- Follower SE inverter is an Storedge inverter. I change the Charge limit and Discharge limit on this inverter which is the one connected to a LG battery. Discharge limit can be modified any time but not the Discharge limit

Could it be related to the status of the leader inverter? Normally, it could be coincidence, I can change the charge limit on the follower inverter when leader inverter status is MPPT and I can't change it when leader status is SLEEPING.

I insist that last winter I could modify this values without problem. Maybe something could have changed (I know many things changed) since last winter version integration or something changed from SolarEdge with some update.

marin849, 2024-10-31:
I have the same issue. I just discovered that I can no longer charge the battery from grid. I have set the inverter to RRCR, Remote, ,"charger from solar power and grid", and "solaredge_i1_storage_charge_limit" is set to 4000W. But the battery does not charge.  AC charge limit is "always allowed". This was working in 3.0.0. I did not change any automations.

FranciscoBigWings, 2024-11-01:
> I have the same issue. I just discovered that I can no longer charge the battery. It is possible to set allowed charge power, but it does not charge. I did not change any automations. I did upgrade the inverter firmware "not so long ago". Will investigate.
> 
> Update: I suspect it is the new firmware. I restored a backup of HAOS with integration verison 3.0.0, and it did not work there either. New firmware is 4.21.23. Last working version was 4.20.36. I will check if it can be rolled back and test again.
> 
> Update2: The inverter can not charge the battery from the grid in Solaredge own "Time of Use" mode or the new manual mode that can be set in the app. It does pass the battery self test and charge and discharge during the self-test.

That was another option, changes on the firmware of the inverter itself. I can write on the field of the “Charge limit” field but after updating the page the value is not set and it appears 0 again. Can you confirm that you can change the Charge limit value and it remains changed?

Can you detail your inverter or inverters configuration? I also suspected that the issue could be related to the leader-follower configuration.

I realised also that is not just the Charge limit that I can’t change at night, there are other values/options that remain changed in HA after the modification.

marin849, 2024-11-01:
> > Can you confirm that you can change the Charge limit value and it remains changed?
Yes.  If there is enough solar surplus the inverter does respect the value and the setting stays.
> Can you detail your inverter or inverters configuration? I also suspected that the issue could be related to the leader-follower configuration.
One single SE10K-RWS with LG Resu 13.
> I realised also that is not just the Charge limit that I can’t change at night, there are other values/options that remain changed in HA after the modification.
I can read and write all values that I have tested.
My problem is that the inverter is unable to charge using grid power. Solaredge support accessed my inverter remotely today and they are also unable to make it charge from the grid. A support case is created to downgrade the firmware to 4.20.36 (the problem started with 4.21.23).

I do not see that this plugin would be the source of my issue.

FranciscoBigWings, 2024-11-02:
Thank you for the confirmation, in my case I can force charge from the grid and modify the Charge limit but only while there is solar power or the inverter is no sleeping.

I think that as well as for you, this might be related to a SolarEdge firmware bug. The fact is that I always used this integration to modify through modbus the Charge limit value. I tried to find the within the inverter Setapp menus the possibility to modify the Charge limit value but it is not possible.

I tried to use Time of use and charge the battery through a profile previously loaded to the inverter but it doesn't work but I suspect that is due the 0W value in the Charge limit. SolarEdge profiles dont allow to set the Charge limit values either so I can't exactly say which is the cause.

Does anyone any way to modify the Charge limit value directly in the inverter? This will help debug my issue

Thanks