WillCodeForCats/solaredge-modbus-multi #708: SE Home Meter stopped working

clabinsky, 2024-12-10:
### Describe the bug

SolarEdge uploaded a firmware update to my energy meter device yesterday (1.0.64 => 1.0.90). I cannot get any values through Modbus anymore since today morning. Any ideas on this?

Device ID: MTR-240-3PC1-D-A-MW
1. https://www.solaredge.com/de/produkte/zaehler-und-sensoren/solaredge-home-energiezaehler

### Expected behavior

A steady flow of sensor values. ;)

### Screenshots

[image]
[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-b9021318c3c4e0163dd38bf0171164c9(5).json](https://github.com/user-attachments/files/18080269/config_entry-solaredge_modbus_multi-b9021318c3c4e0163dd38bf0171164c9.5.json)

### Home Assistant Version

2024.10.0

### solaredge-modbus-multi Version

3.0.2

### Installation Type

Container

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-12-12:
The meter is being identified as a meter, but all of the values the inverter returns for the meter are `0x8000` or `0x0`.

You can try turning everything off and then back on to cold boot the system, but this isn't an issue to fix in the integration. The firmware in the meter or inverter needs to be fixed if a cold boot doesn't help.

clabinsky, 2024-12-12:
The meter started working again 12 hours later without rebooting. Looks like a firmware issue. Thanks for your support anyway!

closed 2024-12-12 by clabinsky