WillCodeForCats/solaredge-modbus-multi #710: The integration is no longer loaded in HA 2024.12.2

Vaszago, 2024-12-12:
### Describe the bug

Since the last upgrade of Homeassistant, the integration is no longer loaded.

[image]

The configuration flow could not be loaded

[image]

### Expected behavior

Not loading

### Screenshots

[image]

The configuration flow could not be loaded

[image]

### Diagnostic File

no file found

### Debug logs

```shell
I didn't find any error message
```

### Home Assistant Version

2024.1.2.2

### solaredge-modbus-multi Version

3.02

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2024-12-12:
2024.12.2 works fine for me.

Does it really not show any log messages? Can you turn on debug logging and reload?

Vaszago, 2024-12-12:
After restarting HA I see logs

Logger: homeassistant.util.loop
Quelle: util/loop.py:77
Erstmals aufgetreten: 07:31:32 (1 Vorkommnisse)
Zuletzt protokolliert: 07:31:32

Detected blocking call to import_module with args ('custom_components.solaredge_modbus_multi',) in /usr/src/homeassistant/homeassistant/loader.py, line 1074: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 708, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 679, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2027, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1026, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)

[Error log.txt](https://github.com/user-attachments/files/18106662/Error.log.txt)

WillCodeForCats, 2024-12-12:
Can you download the full log under Settings, System, Logs

WillCodeForCats, 2024-12-12:
The error log.txt link isn't working.

Vaszago, 2024-12-12:
Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 07:31:32 (1 Vorkommnisse)
Zuletzt protokolliert: 07:31:32

Setup failed for custom integration 'solaredge_modbus_multi': Unable to import component: pymodbus is not installed, or pymodbus version is not supported
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 18, in <module>
    from pymodbus.pdu import ExceptionResponse, ModbusExceptions
ImportError: cannot import name 'ModbusExceptions' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py). Did you mean: 'ModbusException'?

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 18, in <module>
    from .hub import DataUpdateFailed, HubInitFailed, SolarEdgeModbusMultiHub
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 20, in <module>
    raise ImportError("pymodbus is not installed, or pymodbus version is not supported")
ImportError: pymodbus is not installed, or pymodbus version is not supported

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 18, in <module>
    from pymodbus.pdu import ExceptionResponse, ModbusExceptions
ImportError: cannot import name 'ModbusExceptions' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py). Did you mean: 'ModbusException'?

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1026, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 18, in <module>
    from .hub import DataUpdateFailed, HubInitFailed, SolarEdgeModbusMultiHub
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 20, in <module>
    raise ImportError("pymodbus is not installed, or pymodbus version is not supported")
ImportError: pymodbus is not installed, or pymodbus version is not supported

Vaszago, 2024-12-12:
[home-assistant_2024-12-12T06-52-13.540Z.log](https://github.com/user-attachments/files/18106824/home-assistant_2024-12-12T06-52-13.540Z.log)

WillCodeForCats, 2024-12-12:
Thanks for the details, I found the problem.

The issue is that the API for Pymodbus 3.8.0 has changed and my integration was written for the 3.7.x API. This fails to load because it tries to import a class that was removed.

Are you using other modbus integrations?

Vaszago, 2024-12-12:
I use Modbusproxy to retrieve the values ​​from Solaredge.

[https://github.com/Akulatraxas/ha-modbusproxy/](url)

Maybe the error is related to this.

WillCodeForCats, 2024-12-12:
Docker container wouldn't affect other things. It should have to be an integration.

Vaszago, 2024-12-12:
Is the error due to the missing API support.

WillCodeForCats, 2024-12-12:
I can't update the integration for the new API because it will stop working with all existing installations. Home Assistant core is using `pymodbus==3.6.6` so that's what I should support. I've also tested with 3.7.x and that API is compatible.

[Release v3.0.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.0.3) this release adds a requirement for 3.6.6 through 3.7.4. Try upgrading to this version and see if that works.

WillCodeForCats, 2024-12-12:
Now that it's working can you upload the diagnostic file?

Vaszago, 2024-12-12:
[home-assistant_solaredge_modbus_multi_2024-12-12T08-26-28.815Z.log](https://github.com/user-attachments/files/18108030/home-assistant_solaredge_modbus_multi_2024-12-12T08-26-28.815Z.log)
 
Is this the correct file

WillCodeForCats, 2024-12-12:
It's the file from this button in the hub:

[image]

Vaszago, 2024-12-12:
[config_entry-solaredge_modbus_multi-2346e4de85083329484f3296451fdad0.json](https://github.com/user-attachments/files/18108319/config_entry-solaredge_modbus_multi-2346e4de85083329484f3296451fdad0.json)

I hope it fits

Vaszago, 2024-12-12:
Is there a better integration for the optimizers than this integration which I use.

https://github.com/ProudElm/solaredgeoptimizers

WillCodeForCats, 2024-12-12:
Yes, that's the right file. Thanks for the information that helped me quickly fix it.

After looking at this more, it is a mistake I made having requirements incorrectly set without an upper version limit. Pymodbus 3.8.0 introduces breaking API changes that are not backwards compatible. Without an upper limit in the requirements HA will download the latest version of pymodbus which will cause a failure the next time you restart or upgrade Home Assistant.

I've updated the release notes to say [Release v3.0.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.0.3) is a critical update for everyone.

WillCodeForCats, 2024-12-12:
> Is there a better integration for the optimizers than this integration which I use.
> 
> https://github.com/ProudElm/solaredgeoptimizers

Modbus doesn't have optimizer data, so the SolarEdge website is the only way.

WillCodeForCats, 2024-12-13:
Fixed in PR #711

Vaszago, 2024-12-13:
After updating to 2024.12.3 the same error again.
The integration cannot be started.

[home-assistant_2024-12-13T19-45-17.792Z.log](https://github.com/user-attachments/files/18131184/home-assistant_2024-12-13T19-45-17.792Z.log)

WillCodeForCats, 2024-12-13:
The HA version actually does not matter for this issue.

Something is loading incompatible pymodbus version, but I can't tell what that is from the logs.

WillCodeForCats, 2024-12-13:
I'll just make another release and add my own class to copy the one pymodbus removed since I don't know why this is still a problem after fixing the version numbers in the requirements file.

Vaszago, 2024-12-13:
After restarting HA the error did not occur again. I was just shocked that the error occurred again after the restart. That's why I deleted my post when it was working again.

WillCodeForCats, 2024-12-20:
[Release v3.0.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.0.4) removes the dependency on the class the pymodbus removed in the latest versions and replaces it with an internal copy.

Please use this release.

Vaszago, 2024-12-20:
Thanks, the update worked great.

closed 2025-01-01 by WillCodeForCats