WillCodeForCats/solaredge-modbus-multi #721: Stopped working in HA 2025.1.0

fss, 2025-01-04:
### Describe the bug

After migrating the setup to 2025.1.0, integration won't load anymore. here's the error I'm getting:

`Logger: homeassistant.setup
Source: setup.py:334
First occurred: 5:11:41 PM (1 occurrences)
Last logged: 5:11:41 PM

Setup failed for custom integration 'solaredge_modbus': Unable to import component: cannot import name 'ENERGY_KILO_WATT_HOUR' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus/__init__.py", line 17, in <module>
    from .const import (
    ...<7 lines>...
    )
  File "/config/custom_components/solaredge_modbus/const.py", line 22, in <module>
    from homeassistant.const import (
    ...<4 lines>...
    )
ImportError: cannot import name 'ENERGY_KILO_WATT_HOUR' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus/__init__.py", line 17, in <module>
[... 82 lines cut ...]
    ...<7 lines>...
    )
  File "/config/custom_components/solaredge_modbus/const.py", line 22, in <module>
    from homeassistant.const import (
    ...<4 lines>...
    )
ImportError: cannot import name 'ENERGY_KILO_WATT_HOUR' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)`

### Home Assistant Version

2025.1.0

### solaredge-modbus-multi Version

1.1.3

### Installation Type

HAOS

### Read the Instructions

- [X] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [X] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [X] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-01-04:
Hello, you are using a different integration. This is the `solaredge_modbus_multi` integration. Your errors are from using `solaredge_modbus`.

I would recommend removing that integration and changing to solaredge_modbus_multi.

fss, 2025-01-04:
meh, my bad, just followed first found link on GH. of course this issue doesn't belong here

closed 2025-01-04 by fss

WillCodeForCats, 2025-01-04:
Mine might show up first becasue we have more activity and I'm more proactive making updates than the other one is.