WillCodeForCats/solaredge-modbus-multi #732: Only one battery visible

MikeTango, 2025-01-18:
### Describe the bug

Hi,

i recently added a second battery to my setup. After initial fuckups with Solaredge being unable to upgrade from the delivery firmware (1.12) to the current one, they made it work but it identifies only as one battery with twice the capacity. I am pretty sure that is intentional, but is it possible to get data for both batteries seperately via modbus? "Auto-discover batteries" yields no results. Screenshot with firmwares attached. 

### Expected behavior

Seeing two batteries

### Screenshots

[image]

### Diagnostic File

none

### Debug logs

```shell

```

### Home Assistant Version

2025.1.1 

### solaredge-modbus-multi Version

3.0.4

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-01-18:
It is possible, but only if the inverter exports the batteries over modbus. Up to 3 batteries are supported.

I don't personally have batteries so I can't offer any help on specific battery configurations.

MikeTango, 2025-01-18:
Well, the home assistant only sees one battery via modbus TCP, is there a way to dump the whole modbus register to see how it formats the battery?

WillCodeForCats, 2025-01-18:
Yes, turn on debug logging in the device panel.

WillCodeForCats, 2025-01-19:
[image]

MikeTango, 2025-01-19:
`
2025-01-19 08:29:10.102 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 86 86
2025-01-19 08:29:10.103 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_MaxChargePower 0x459c4000 <class 'float'>
2025-01-19 08:29:10.103 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_MaxDischargePower 0x459c4000 <class 'float'>
`

There are no multiple batteries. I tried the auto-discover but i now am pretty sure the inverter only hands over one battery register that it calculates from the two batteries.

Then i downloaded the diagnosis data from the other dropdown menu, pardon my german:

[image]

There is only a B1. While toying with that, i think i found an unrelated bug:

There are two instances in which the serial numbers do not get redacted in the json. One is in the Battery block:

`    "battery_id_1": {
      "device_info": {
        "identifiers": "**REDACTED**",
        "name": "Solaredge I1 B1",
        "manufacturer": "SolarEdge",
        "model": "SolarEdge Home Battery 9.2kWh",
        "serial_number": "**REDACTED**",
        "sw_version": "48V DCDC 3.5.6",
        "via_device": [
          "solaredge_modbus_multi",
          "INVERTER SERIAL NUMBER PLAINTEXT HERE"
        ]
      },`

One is in the Meter data

`    "meter_id_1": {
      "device_info": {
        "identifiers": "**REDACTED**",
        "name": "Solaredge I1 M1",
        "manufacturer": "SolarEdge",
        "model": "MTR-240-3PC1-D-A-MW",
        "serial_number": "**REDACTED**",
        "sw_version": "1.0.60",
        "hw_version": "Export+Import",
        "via_device": [
          "solaredge_modbus_multi",
          "INVERTER SERIAL NUMBER PLAINTEXT HERE"
        ]
      },`

regards, 

MT

cyberkryten, 2025-01-31:
I have three batteries and each are shown individually.  

On my own ($%^&ed up installation) they struggled to figure out how to connect three batteries, bricked one, replaced it, failed to upgrade firmware, bricked the inverter, replaced it and then I got three batteries but they appear as B1.1, B1.2 and B1.4 in the SE Web Portal.

What do you see on the SE App and SE Web Portal?

WillCodeForCats, 2025-02-01:
> There are two instances in which the serial numbers do not get redacted in the json. One is in the Battery block:

Fixed in PR [#743](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/743)

closed 2025-02-27 by WillCodeForCats