WillCodeForCats/solaredge-modbus-multi #733: Receive no data from Meter - but it is automatically detected

ScherzerThomas, 2025-01-18:
### Describe the bug

Hi,

i've tried to switch from solaredge-modbus to this integration, because it has several connection problems since the last month.
(It always worked for a few days, then i had to power off-on the inverter, so that a modbus connection was reestablished - but ignore this)

The problem is, that i get no data from the meter (it measures export+import), the battery and inverter entities work all fine.

The  inverter and the meter (1.0.90) seems to have the latest firmware version (did it update itself? - i didn't do anything)

With this problem i've to go back to the other integration - but i don't want due to the connection issues.

Core 2025.1.2
Supervisor 2024.12.3
Operating System 14.1
Frontend 20250109.0

### Expected behavior

(Useful) values for the meter. All values apart from the diagnostics are undefined.

### Screenshots

Devices - these are correct
[image]

Useful values for the inverter
[image]

Useful values for the battery
[image]

**Unavailable entities of the meter**
[image]

Meter version
[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-01JHXCF0QHANDFR3HK9FJGR7KM.json](https://github.com/user-attachments/files/18465817/config_entry-solaredge_modbus_multi-01JHXCF0QHANDFR3HK9FJGR7KM.json)

### Debug logs

```shell
This is what i get, when i turn on diagnostic logging

2025-01-18 20:14:29.011 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Disconnectng from 192.168.1.230:1502 (clear_client=False).
2025-01-18 20:14:29.011 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.182 seconds (success: True)
2025-01-18 20:14:38.829 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Retry limit=5 time=800 ratio=3
2025-01-18 20:14:38.829 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Connecting to 192.168.1.230:1502 ...
2025-01-18 20:14:38.834 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 10.8
2025-01-18 20:14:38.865 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 16 16
2025-01-18 20:14:38.886 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 40 40
2025-01-18 20:14:38.955 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 2 2
2025-01-18 20:14:38.955 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_DID 0x67 <class 'int'>
2025-01-18 20:14:38.956 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: C_SunSpec_Length 0x32 <class 'int'>
2025-01-18 20:14:38.956 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current 0x33d <class 'int'>
2025-01-18 20:14:38.956 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_A 0x114 <class 'int'>
2025-01-18 20:14:38.956 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_B 0x113 <class 'int'>
2025-01-18 20:14:38.957 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_C 0x114 <class 'int'>
2025-01-18 20:14:38.957 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Current_SF -0x2 <class 'int'>
2025-01-18 20:14:38.957 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_AB 0xf55 <class 'int'>
2025-01-18 20:14:38.957 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_BC 0xf39 <class 'int'>
2025-01-18 20:14:38.958 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_CA 0xf49 <class 'int'>
2025-01-18 20:14:38.958 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_AN 0x8e0 <class 'int'>
2025-01-18 20:14:38.958 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_BN 0x8db <class 'int'>
2025-01-18 20:14:38.958 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_CN 0x8bc <class 'int'>
2025-01-18 20:14:38.958 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: AC_Voltage_SF -0x1 <class 'int'>
[... 132 lines cut ...]
2025-01-18 20:14:39.038 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor6 0x0 <class 'int'>
2025-01-18 20:14:39.038 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor7 0x0 <class 'int'>
2025-01-18 20:14:39.038 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1B1: B_Event_Log_Vendor8 0x0 <class 'int'>
2025-01-18 20:14:39.038 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Disconnectng from 192.168.1.230:1502 (clear_client=False).
2025-01-18 20:14:39.038 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.209 seconds (success: True)
2025-01-18 20:14:48.829 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Retry limit=5 time=800 ratio=3
```

### Home Assistant Version

2025.1.2

### solaredge-modbus-multi Version

3.0.4

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-01-18:
You can try power cycling everything to restart communication, but the data is correct even if it is zero.

WillCodeForCats, 2025-01-18:
Also see: discussion https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/686

ScherzerThomas, 2025-01-18:
I've tried to switch back to the other integration (solaredge_modbus) - it also doesn't work any more so this must be a further problem

Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 244, in _refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 320, in read_modbus_data
    and self.read_modbus_data_meter1()
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 331, in read_modbus_data_meter1
    return self.read_modbus_data_meter("m1_", 40190)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 480, in read_modbus_data_meter
    exported = validate(self.calculate_value(exported, energywsf), ">", 0)
  File "/config/custom_components/solaredge_modbus/__init__.py", line 160, in validate
    raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)

ScherzerThomas, 2025-01-18:
> Also see: discussion [#686](https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/686)

Actually, this (https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/686#discussioncomment-10899586) was the solution -  now it runs - I'll check if it remains stable for longer time.
Nevertheless, thanks for the quick solution.

I think you could emphasize this issue (including a graceful turn off-on procedure) in the FAQs, as i'm sure this happens more often.

WillCodeForCats, 2025-01-18:
I've added it to the Known Issues page in the wiki: https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues

closed 2025-01-18 by ScherzerThomas