WillCodeForCats/solaredge-modbus-multi #739: I moved inverter connectivity from WiFI to LAN and now modBUS is not connecting anymore

nicola-spreafico, 2025-01-24:
### Describe the bug

Hello,
I understand this might no be an issue of the integration itself, but given the type of problem I might found someone here that can help me understand what is going on.

Long story short... One month ago I connected the inverter via WiFi, I enabled the modBUS and I was able to connect this integration without additional issue. Due some connectivity issue (my home wifi is not very stable), I connected the inverter via Ethernet to rely on a more stable connection. Once I changed the connection I made the inverter "Forget" the WiFI credentials, and now the modBUS integration is not working anymore (I corrected the IP with the new Ethernet one).

Full details here: https://community.home-assistant.io/t/solaredge-modbus-multi-switching-from-wifi-to-lan-doesnt-connect-anymore/835229

### Expected behavior

modBUS connection should work with LAN connectivity.

As you can see from the screenshots, the inverter LAN IP is set to 192.168.50.243

I tried to use an external modBUS client and the connections works:
[image]
It says "connected".

At this point I'm not sure if the issue is the inverter or the integration

### Screenshots

Right now I assigned static IP to both interfaces (WiFi or LAN) and neither are working:

[image]
[image]

You may find Solaredge SetApp screenshot in the Home Assistant forum

### Diagnostic File

```
2025-01-24 21:07:06.892 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-01-24 21:07:06.892 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2025-01-24 21:07:06.892 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.005 seconds (success: False)
2025-01-24 21:07:14.466 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: inverter_list=[1], detect_meters=True, detect_batteries=False, detect_extras=False, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
2025-01-24 21:07:14.466 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] pymodbus version 3.7.4
2025-01-24 21:07:14.466 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Retry limit=5 time=800 ratio=3
2025-01-24 21:07:14.466 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] New AsyncModbusTcpClient: reconnect_delay=0 reconnect_delay_max=3 timeout=3
2025-01-24 21:07:14.466 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Connecting to 192.168.50.243:1502 ...
2025-01-24 21:07:17.471 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-01-24 21:07:17.471 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2025-01-24 21:07:17.471 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.005 seconds (success: False)
2025-01-24 21:07:35.285 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: inverter_list=[2], detect_meters=True, detect_batteries=False, detect_extras=False, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
2025-01-24 21:07:35.285 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] pymodbus version 3.7.4
2025-01-24 21:07:35.285 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Retry limit=5 time=800 ratio=3
2025-01-24 21:07:35.285 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] New AsyncModbusTcpClient: reconnect_delay=0 reconnect_delay_max=3 timeout=3
2025-01-24 21:07:35.285 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Connecting to 192.168.50.240:1502 ...
2025-01-24 21:07:38.290 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-01-24 21:07:38.290 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 21.6
2025-01-24 21:07:38.290 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 3.005 seconds (success: False)
2025-01-24 21:08:11.924 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: inverter_list=[1], detect_meters=True, detect_batteries=False, detect_extras=False, keep_modbus_open=False, adv_storage_control=False, adv_site_limit_control=False, allow_battery_energy_reset=False, sleep_after_write=0, battery_rating_adjust=0, 
2025-01-24 21:08:11.924 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] pymodbus version 3.7.4
2025-01-24 21:08:11.924 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Retry limit=5 time=800 ratio=3
2025-01-24 21:08:11.924 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] New AsyncModbusTcpClient: reconnect_delay=0 reconnect_delay_max=3 timeout=3
2025-01-24 21:08:11.924 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Connecting to 192.168.50.241:1502 ...
```

### Debug logs

```shell

```

### Home Assistant Version

2025.1.3

### solaredge-modbus-multi Version

3.0.5

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-01-27:
I'm not aware of any modbus port difference between wifi and ethernet, but my inverters never had wifi support so I don't have anything to compare to.

Make sure that modbus is still enabled after changing to ethernet.

WillCodeForCats, 2025-01-27:
Have you tried doing a cold restart of your inverter after removing the wifi configuration?

1. Turn P/1/0 switch to 0 and wait until green LED starts blinking
2. AC off
3. Check that LEDs are off
4. AC on
5. P/1/0 switch back to 1

cyberkryten, 2025-01-31:
What IP are you expecting?  Your test showing it works was to 192.168.50.243 but you are getting failures to 192.168.50.48 so which is correct?  There are also attempts on .240 and .241 so I'm not sure what IPs you have setup?

Does the test to .48 work?

nicola-spreafico, 2025-01-31:
Sorry about the confusion, after many experiments I did wifi yes/no and ethernet yes/no it seems I gave you some misleading screenshots. Tomorrow I will do a recap with the information I gathered so far.

cyberkryten, 2025-02-01:
It's worth trying a port scan on the Inverter to see what ports are open - that will tell you something useful as if the port isn't open then you'll probably have to speak to your installer or SE Support.

1502 isn't included by default, so you won't see it open - all I get is 8181 so I have to scan it specifically but it does show as open on mine - here's what I'd expect to see for your example:

```
~ nmap -q -p 1502 192.168.50.48
Starting Nmap 7.95 ( https://nmap.org ) at 2025-02-01 12:19 GMT
Nmap scan report for 192.168.50.48
Host is up (0.00094s latency).

PORT     STATE SERVICE
1502/tcp open  shivadiscovery
MAC Address: xx:xx:xx:xx:xx:xx (SolarEdge Technologies)

Nmap done: 1 IP address (1 host up) scanned in 0.11 seconds
```