WillCodeForCats/solaredge-modbus-multi #742: Mehrere solaredge_modbus_multi-Konfigurationseinträge mit derselben eindeutigen ID

LCB3D, 2025-01-28:
### Describe the bug

Seit dem letzten Updatet erscheint bei mir der Fehler und ich habe keine Idee mehr wo ich das beheben kann.
Es gibt mehrere solaredge_modbus_multi-Konfigurationseinträge mit derselben eindeutigen ID.
Die Konfigurationseinträge heißen 'SolarEdge 10K', 'SolarEdge 5000'

Um diesen Fehler zu beheben, [konfiguriere die Integration](http://homeassistant.local:8123/config/integrations/integration/solaredge_modbus_multi) und entferne alle bis auf eines der Duplikate.

Hinweis: Nach dem Entfernen dieser Duplikate kann eine weitere Gruppe von Duplikaten sichtbar werden.

### Expected behavior

Bisher hat alles reibungslos funktioniert, nun tritt der Fehler auf, woran liegt das?

### Screenshots

[image]

### Diagnostic File

[home-assistant_2025-01-28T18-06-18.153Z.log](https://github.com/user-attachments/files/18577583/home-assistant_2025-01-28T18-06-18.153Z.log)

### Debug logs

```shell

```

### Home Assistant Version

14.2 2025.1.4

### solaredge-modbus-multi Version

3.0.6

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-01-28:
Upload the diagnostic file.

[image]

LCB3D, 2025-01-28:
[image]

[image]

[config_entry-solaredge_modbus_multi-01JG0VF15WG0ZA3KG9FT2P7TKM.json](https://github.com/user-attachments/files/18577857/config_entry-solaredge_modbus_multi-01JG0VF15WG0ZA3KG9FT2P7TKM.json)

[config_entry-solaredge_modbus_multi-01JG0VHMJZ2SGK0261XYCEP9JZ.json](https://github.com/user-attachments/files/18577861/config_entry-solaredge_modbus_multi-01JG0VHMJZ2SGK0261XYCEP9JZ.json)

WillCodeForCats, 2025-01-30:
The IP address + port are the same in both hubs, so that is why there is a duplicate error.

You can use only one hub and set the device ID list to `1,3` instead of having two hubs.

What probably happened is sometime in the past they had different IP addresses to add (there was always a duplicate IP check) but changing the address through reconfigure didn't check for duplicates. Now the configuration upgrade to internal config 2.1 fixes all of the unique ID to match the host + port, reconfigure and repair also.

> Google Translate

Die IP-Adresse und der Port sind in beiden Hubs gleich, daher liegt ein doppelter Fehler vor.

Sie können nur einen Hub verwenden und die Geräte-ID-Liste auf `1,3` setzen, anstatt zwei Hubs zu haben.

Was wahrscheinlich passiert ist, ist, dass sie irgendwann in der Vergangenheit verschiedene IP-Adressen hinzufügen mussten (es gab immer eine Prüfung auf doppelte IPs), aber beim Ändern der Adresse durch Neukonfigurieren wurde nicht nach Duplikaten gesucht. Durch das Konfigurations-Upgrade auf die interne Konfiguration 2.1 werden nun alle eindeutigen IDs so korrigiert, dass sie mit dem Host und dem Port übereinstimmen. Außerdem werden sie neu konfiguriert und repariert.

closed 2025-02-01 by WillCodeForCats

LCB3D, 2025-02-01:
Unfortunately, since configuring with a hub, I keep having problems with the configuration not working or all entities not being available. what is the reason for this?

[image]

[config_entry-solaredge_modbus_multi-01JG0VF15WG0ZA3KG9FT2P7TKM.json](https://github.com/user-attachments/files/18629667/config_entry-solaredge_modbus_multi-01JG0VF15WG0ZA3KG9FT2P7TKM.json)