WillCodeForCats/solaredge-modbus-multi #744: SolarEdge batteries connected via EnergyNet fails initialization

aarondrabeck, 2025-01-29:
### Describe the bug

When I turn on battery auto-detect, the plugin fails to initialize.  

[image]

Is this a known issue where modbus cannot see batteries when connected via wireless Energy Net or is something else going wrong?

```
2025-01-29 10:48:50.946 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Found I1M2
2025-01-29 10:48:50.946 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for meter I1M3
2025-01-29 10:48:50.977 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 67 67
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: C_SunSpec_DID 0x2bd <class 'int'>
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: C_SunSpec_Length 0x99 <class 'int'>
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: C_Manufacturer 
d	 <class 'str'>
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: C_Model i <class 'str'>
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: C_Option  <class 'str'>
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: C_Version  <class 'str'>
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: C_SerialNumber  <class 'str'>
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: C_Device_address 0x8000 <class 'int'>
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M3: Meter 3 ident incorrect or not installed.
2025-01-29 10:48:50.978 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for battery I1B1
2025-01-29 10:49:02.980 ERROR (MainThread) [pymodbus.logging] No response received after 3 retries, continue with next request
2025-01-29 10:49:02.981 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Disconnectng from 10.2.0.100:1502 (clear_client=False).
2025-01-29 10:49:02.982 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 12.233 seconds (success: False)

### Expected behavior

Gracefully fail to detect battery and still load the plugin 

### Diagnostic File

n/a

### Debug logs

```shell

```

### Home Assistant Version

2025.1.4

### solaredge-modbus-multi Version

3.0.6

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-01-29:
Wireless is known to be unstable, unfortunately. You can try power cycling the inverter:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues#missing-data-on-meters-with-energynet-or-home-network

Also try turning off "Auto-Detect Additional Entities" if it's enabled.

WillCodeForCats, 2025-01-29:
Also, since the inverter stopped responding at batteries, you might also need to power cycle those. I don't personally have batteries to know what the steps for that would be, though.

If you try turning off "Auto-Detect Batteries" that means battery communication is the problem.

All of these options are under the Configure button in the device hub.

WillCodeForCats, 2025-01-29:
Oops, sorry I misunderstood you were trying to turn on battery auto detect, not that it was already on.

If it returned ExceptionResponse that usually means the inverter is saying the request is not supported, so its not exporting battery access over modbus. You can still try the power cycle stuff though.

aarondrabeck, 2025-01-30:
I was able to hardwire for testing however there seems to be no effective change.  I noticed, with the latest SetApp/Firmware, SolarEdge batteries are no longer added in the Site Comm -> rs485-1 menu (unless you have a LG battery) regardless hardwired or wireless.  In fact, the batteries do not even appear under rs485-1 when commissioned.  Instead, they are in the Device Manager section, which is where the Energy Net devices appear.  

Perhaps there is some type of isolation in these comm bus's and battery communication isn't exposed over modbus at all.

Is there anyone else out there with a SolarEdge Home Battery Bank + Home Hub Invertor who is able to see battery data over ModBus?

cyberkryten, 2025-01-31:
> Wireless is known to be unstable, unfortunately. You can try power cycling the inverter:

Unfortunately this is wrong for SE, at least some SE systems where more than one battery is installed.  Cabled RS485 responses are too slow and the Inverter fails to control the batteries when physically connected leading to import/export issues.  

The SE recommended method of 'connection' is Wireless to the batteries and wired to the Backup Gateway (if present).

Doesn't make sense as we all prefer wired connections but there's something very wrong with their RS485 over Ethernet stack.

WillCodeForCats, 2025-02-01:
Sometimes SolarEdge has updated their firmware to add modbus support to wireless connected devices. The wireless meters were hidden from modbus until recently ([Discussion #686](https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/686) and [Issue #252](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/252)). You could try reporting to SolarEdge support.

peter-dolkens, 2025-02-05:
I've got a SolarEdge Home Battery Bank + Home Hub Inverter and Modbus had been working perfectly fine until the most recent update to solaredge-modbus-multi

I'm tempted to roll back, but I also had installed predbat in monitoring-only mode recently, so have uninstalled that for now and am waiting to see if things will be more stable now that I've made that change

WillCodeForCats, 2025-02-05:
All data is collected through the inverter like a proxy. The integration can't directly communicate with batteries or meters.

WillCodeForCats, 2025-02-05:
Also upload your diagnostic files at the time a problem is occurring: screenshots don't show all the the data the integration has from the inverter.

WillCodeForCats, 2025-02-05:
In the past I have seen the inverter report all zeros for battery values.

aarondrabeck, 2025-02-12:
I have been getting the flaky data issue too for a year or more, prior to the battery install.  It almost felt like fail-to-ban is installed on the inverter.  I tried adjusting time / persist connection, to no avail.  Sometimes hard reboots of HA would a solve it, sometimes not.  I could never narrow it down.  

I will keep trying ModBus as new firmware is released.  

I have asked SolarEdge about the possibility of adding a local API to their devices, not sure that will go any where though but it would be ideal.  

For now I was able to query SolarEdge's V2 API and it has been very reliable, although the resolution is only 5 mins.  You must have a installer account at the moment to gain access, at least for now.

Here is my config for that incase anyone wants to give it a try.
```  
rest:
  - scan_interval: 300
    resource: "https://monitoringapi.solaredge.com/v2/sites/3581845/power-flow"
    headers:
      X-API-Key: "***"
      X-Account-Key: "***"
    sensor:
      - name: "SolarEdge Site API PV Power"
        value_template: "{{ value_json['pv']['power'] }}"
        device_class: power
        unit_of_measurement: "kW"
      - name: "SolarEdge Site API PV Power Status"
        value_template: "{{ value_json['pv']['active'] }}"
      - name: "SolarEdge Site API Home Consumption Power"
        value_template: "{{ value_json['load']['power'] }}"
        device_class: power
        unit_of_measurement: "kW"
      - name: "SolarEdge Site API Home Consumption  Active"
        value_template: "{{ value_json['load']['active'] }}"
      - name: "SolarEdge Site API Battery Power"
        value_template:
          "{% if value_json['storage']['status'] == 'DISCHARGE'  %}
          {{ value_json['storage']['power']|float(0) }}
          {% else %}
          {{ value_json['storage']['power']|float(0) * -1 }}
          {% endif %}"
        device_class: power
        unit_of_measurement: "kW"
      - name: "SolarEdge Site API Battery  Active"
        value_template: "{{ value_json['storage']['active'] }}"
      - name: "SolarEdge Site API Battery Status"
        value_template: "{{ value_json['storage']['status'] }}"
      - name: "SolarEdge Site API Battery Charge Level"
        value_template: "{{ value_json['storage']['chargeLevel'] }}"
        device_class: battery
        unit_of_measurement: "%"
      - name: "SolarEdge Site API Grid Power"
        value_template: "{% if value_json['grid']['status'] == 'IMPORT'  %}
          {{ value_json['grid']['power']|float(0) }}
          {% else %}
          {{ value_json['grid']['power']|float(0) * -1 }}
          {% endif %}"
        device_class: power
        unit_of_measurement: "kW"
      - name: "SolarEdge Site API Grid Active"
        value_template: "{{ value_json['grid']['active'] }}"
      - name: "SolarEdge Site API Grid Status"
        value_template: "{{ value_json['grid']['status'] }}"
  - scan_interval: 120
    resource: "https://monitoringapi.solaredge.com/v2/sites/3581845/overview"
    headers:
      X-API-Key: "***"
      X-Account-Key: "***"
    sensor:
      - name: "Total PV production power"
        unique_id: sensor.solaredge_api_total_pv_production_power
        value_template: "{{ value_json['production']['total'] }}"
        device_class: energy
        unit_of_measurement: "Wh"
        state_class: total_increasing
      - name: "Total PV production to home power"
        unique_id: sensor.solaredge_api_total_pv_production_to_home_power
        value_template: "{{ value_json['production']['toSelfConsumption'] }}"
        device_class: energy
        unit_of_measurement: "Wh"
        state_class: total_increasing
      - name: "Total PV production to battery power"
        unique_id: sensor.solaredge_api_total_pv_production_to_battery_power
        value_template: "{{ value_json['production']['toStorage'] }}"
        device_class: energy
        unit_of_measurement: "Wh"
        state_class: total_increasing
      - name: "Total PV production to grid power"
        unique_id: sensor.solaredge_api_total_pv_production_to_grid_power
        value_template: "{{ value_json['production']['toGrid'] }}"
        device_class: energy
        unit_of_measurement: "Wh"
        state_class: total_increasing
      - name: "Total from battery consumption"
        unique_id: sensor.solaredge_api_total_from_battery_consumption
        value_template: "{{ value_json['consumption']['fromStorage'] }}"
        device_class: energy
        unit_of_measurement: "Wh"
        state_class: total_increasing
      - name: "Total from grid consumption"
        unique_id: sensor.solaredge_api_total_from_grid_consumption
        value_template: "{{ value_json['consumption']['fromGrid'] }}"
        device_class: energy
        unit_of_measurement: "Wh"
        state_class: total_increasing

```
[image]

closed 2025-02-27 by WillCodeForCats