WillCodeForCats/solaredge-modbus-multi #745: Several essential sensors „unavailable“

w1Ngx, 2025-01-31:
### Describe the bug

I freshly installed this addon and based some energy dashboards on it. After several variables showed a permanent 0 i checked the sensors and noticed that 10+ are unavailable (e.g. sensor.solar_panel_production_kwh or sensor.solar_grid_to_battery_kwh

### Expected behavior

All sensors are available and show values.

### Screenshots

[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-01JJMWAJMR24JNQ13YPG17DBHE.json](https://github.com/user-attachments/files/18614017/config_entry-solaredge_modbus_multi-01JJMWAJMR24JNQ13YPG17DBHE.json)

### Debug logs

```shell

```

### Home Assistant Version

2025.1.4

### solaredge-modbus-multi Version

3.0.6

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-01-31:
Those are template sensors, not from the integration.

w1Ngx, 2025-01-31:
Hm, where do they come from and where do i see which sensor is providing them with data?

WillCodeForCats, 2025-01-31:
They might be in the `template.yaml` file or under Helpers.

w1Ngx, 2025-01-31:
I removed now all these additional templates and exactly copy pasted your template for 1 inverter 1 battery. After that i deleted all template sensors, which were „not provided“, so it should be clean now. However when looking at the tesla card everything is showing 0 and the same for the standard Energy Dashboard from HA.

Is there something else to consider?

cyberkryten, 2025-01-31:
Go to Settings->Integrations->SolarEdge Modbus Multi->Devices

Then select your Inverter, Meter, Battery(or batteries) and look at the data there - if you have various sensors showing data that changes then this integration is working fine.

It could be you have named things differently, copied something wrong or have other issues in your templates or config but you'll have to investigate the problems with the Template Sensors or the Tesla Card configm but it isn't an integration issue.

For example:

[image]

w1Ngx, 2025-01-31:
Ok i will try this configuration for the energy dashboard.
How do you feed the Tesla card without template sensors? can you show which entities you use?

WillCodeForCats, 2025-01-31:
I don't use the Tesla card.

cyberkryten, 2025-02-01:
In all honesty, if you're not a Dev and understand programming logic then I think getting the Tesla card working is pretty much impossible as every system is slightly different.

I've been using it for six months and I'm still finding issues with the config and scenarios of solar/import/export/discharge where one of the calculations is wrong.  Also, firmware updates from SE have changed figures it reports and one in November meant their own portal shows zero solar energy generated since then!

You need to decide how you want to categorise the usage and calculate what energy is sent to where.  For example, if the house is drawing 1.5kW, you are exporting 3.5kW whilst generating 1kW of solar is that solar going to the house or to the grid?

The inverter doesn't tell you - it just knows it is getting 1kW from solar, 4kW from batteries and outputting 5kW - how you split that up for the Tesla Card using template sensors is your decision.

cyberkryten, 2025-02-01:
So, here's my Tesla Card config and this shows why you can't just take someone else's config and will have to create it yourself because your setup is unique and will differ from mine.  You need to setup template sensors to do the calculations, based on the examples but relating to the data you have available.

I've got various calculated sensors.  Some data comes from the SE Modbus values, some from the batteries (it needs one figure so I add three together), the Grid data comes from a Hildebrand Glow to read the Smart Meter DCC values directly (as that is more accurate), Load values come from a combination of Shelly CT clamps and Emporia energy CT.

I also have appliance figures for the Zappi EV Charger as well as other Shelly CT clamps for the Aircon system.

```
          - type: custom:tesla-style-solar-power-card
            view_layout:
              grid-area: Flow
            name: New Power Flow
            threshold_in_k: 0
            house_without_appliances_values: 1
            change_house_bubble_color_with_flow: 1
            generation_entity: sensor.semodbus_power_solar_generation_alt
            generation_extra_entity: sensor.solcast_pv_forecast_power_now_kw
            generation_to_grid_entity: sensor.semodbus_power_pv_grid
            generation_to_house_entity: sensor.semodbus_power_pv_load
            generation_to_battery_entity: sensor.semodbus_power_pv_battery
            grid_to_house_entity: sensor.semodbus_power_grid_load
            grid_to_battery_entity: sensor.semodbus_power_grid_battery
            grid_entity: sensor.smart_meter_electricity_power
            grid_extra_entity: sensor.smart_meter_electricity_power
            battery_to_house_entity: sensor.semodbus_power_battery_load
            battery_to_grid_entity: sensor.semodbus_power_battery_grid
            battery_entity: sensor.semodbus_power_battery_chargedischarge
            battery_extra_entity: sensor.calc_battery_all_state
            appliance1_consumption_entity: sensor.myenergi_zappi_21760455_power_ct_internal_load
            appliance2_consumption_entity: sensor.calc_total_aircon_power_shellyct
            appliance2_icon: mdi:snowflake-thermometer
            house_entity: sensor.semodbus_power_total_load
            grid_icon: mdi:transmission-tower
            generation_icon: mdi:solar-panel-large
            house_icon: mdi:home
```

I would add that the SE Modbus values cause display issues in the card because of the lag between DC sources - the battery charge/discharge start/stop is not instant (I have three) which causes massive swings.  Once you add batteries, the SE Inverter no longer provides a 'Solar' figure itself on Modbus (at least, mine doesn't) just Battery DC and Total DC (no idea why, but maybe they gave up because of these inaccuracies!) - this means I have to calculate it.  

Using a simple DC +/- DC calculation often shows a Solar Generation figure on cloudy wintertime days which is above the maximum panel output in the peak of summer.  

Because of this, it also shows generation during the night - that alone causes any calculated sensors used elsewhere (eg on the Tesla card) to be wrong a lot of the time.

As a result, I have implemented smoothing caps on increases and nighttime checks - this is all BEFORE adding in the various template sensors to try to split up the power and allocate it to sources and destinations!

```
# Use Triggers so the Generation figure only updates when the key values update:
#   The Total Calculated DC value of the three batteries (B1+B2+B3) is updated
#     This avoids the issue where one battery has updated but the total has not
#   The Inverter DC Power changes
#   Every Minute - failsafe to force updates when sensors stop updating
#
# If Battery DC value is positive then it is being charged, negative is discharging
# Negative Inverter value, means Inverter is charging batteries from Grid
#   Subtract negated InverterDC value from Battery DC to get non-grid amount
# Positive Inverter value means the Inverter is providing power to the house
#   That can only come from Batteries or Solar so add both DC values together
#
# Modifiers Applied:
#   Allow Generation from Dawn to Sunset - at night will always be zero
#   Limit each increase or decrease to a set percentage of the previous value
#   Cap at Low end Limit if previous value was zero
#   Hard Cap the Max Generation value at the theoretical total of panels
#   When less than 10% of LowVal, allow drop to zero
- trigger:
    - platform: state
      entity_id:
        - sensor.semodbus_power_batteries_dc_power
        - sensor.solaredgemodbus_i1_dc_power
    - platform: time_pattern
      minutes: /1
[... 19 lines cut ...]
          {% if InvDC < 0 %}
            {% if BattDC > (-InvDC) %}
              {% set myvalue = min ((BattDC - (-InvDC)), MaxPanels) %}          
            {% else %}
              {% set myvalue = 0 %}
            {% endif %}
          {% else %}
            {% set myvalue = min ((InvDC + BattDC), MaxPanels) %}          
            {% if myvalue < 0 %}
              {% set myvalue = 0 %}
            {% endif %}
          {% endif %}        

          {% if myvalue > LastVal and LastVal == 0 %}
            {% set myvalue = min([myvalue , LowVal]) %}            
          {% elif myvalue > LastVal %}
            {% set myvalue = min([(min([myvalue , LastVal * PCTInc])), MaxPanels]) %}
          {% elif myvalue < LastVal and LastVal >= (LowVal/10) %}
            {% set myvalue = max([myvalue , LastVal * PCTDec]) %}                
          {% endif %}

        {% endif %}        
        {{ myvalue | round(1) }}

```