WillCodeForCats/solaredge-modbus-multi #747: Unexpected error fetching SolarEdge Coordinator data

anselmhuewe, 2025-02-01:
### Describe the bug

Hi,
since a couple of days, my SolarEdge modbus integration reports an error. It may have been caused by an update of the ha core system, but not sure.
Reloading the ingegration or restarting ha does not help.

**Error message:**

[image]

**Error report from protocols:**

Can anybody help?

### Expected behavior

Integration shall start.

### Diagnostic File

-

### Debug logs

```shell
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solaredge_modbus_multi
Quelle: helpers/update_coordinator.py:379
Integration: SolarEdge Modbus Multi (Dokumentation, Probleme)
Erstmals aufgetreten: 16:05:27 (12 Vorkommnisse)
Zuletzt protokolliert: 16:10:21

Unexpected error fetching SolarEdge Coordinator data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 379, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 190, in _async_update_data
    return await self._refresh_modbus_data_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<10 lines>...
    )
    ^
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 234, in _refresh_modbus_data_with_retry
    raise ex
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 231, in _refresh_modbus_data_with_retry
    return await self._hub.async_refresh_modbus_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 364, in async_refresh_modbus_data
    await self.connect()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 470, in connect
    self._client = AsyncModbusTcpClient(
                   ~~~~~~~~~~~~~~~~~~~~^
        host=self._host,
        ^^^^^^^^^^^^^^^^
    ...<4 lines>...
        timeout=self._mb_timeout,
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
TypeError: AsyncModbusTcpClient.__init__() got an unexpected keyword argument 'retry_on_empty'
```

### Home Assistant Version

2025.1.4

### solaredge-modbus-multi Version

2.4.15

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-02-01:
2.4.15 is too old to use with HA 2025.1.4

The issue you are reporting was fixed in [Release v2.4.19](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.4.19).

The current version is [Release v3.0.6](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.0.6).

closed 2025-02-01 by WillCodeForCats

WillCodeForCats, 2025-02-01:
Upgrade to the latest version of the integration.

anselmhuewe, 2025-02-03:
thanks for your fast help. That was the mistake.