WillCodeForCats/solaredge-modbus-multi #755: Select function not working, storage command mode

frank8the9tank, 2025-02-06:
### Describe the bug

When using a StorEdge inverter, when selecting an option from the select function it gives an error:
Failed to perform the action select/selection_option. object of type "int" has no len()

[image]

Have tries restarting HA a couple of times.
Disabling and rearranging
Removing the integration  and adding it back again.
Have tried version 3.0.6,, 3.0.5, and 3.0.4 all having the same issue for me.

I use these control in a couple of automation to control the charing and discharging function of the inverter and connected batteries.
Have used this integration for a long time already, and don't known why this is giving trouble now.
When selecting a value for this select using a automation also does not work

### Expected behavior

Have the control select the correct function for the inverter.

### Diagnostic File

Logger: homeassistant.components.automation.storedge_battery_control_control_charge_and_dischage_based_on_setpoint_v2
Source: helpers/script.py:526
integration: Automation (documentation, issues)
First occurred: 12:50:48 (17493 occurrences)
Last logged: 16:05:54

Storedge - Battery control - Control charge and dischage based on setpoint (input number): If at step 1: If at step 1: Error executing script. Unexpected error for device at pos 1: object of type 'int' has no len()
Storedge - Battery control - Control charge and dischage based on setpoint (input number): If at step 1: Error executing script. Unexpected error for if at pos 1: object of type 'int' has no len()
Storedge - Battery control - Control charge and dischage based on setpoint (input number): Error executing script. Unexpected error for if at pos 1: object of type 'int' has no len()
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 779, in _async_device_step
    await device_action.async_call_action_from_config(
        self._hass, self._action, self._variables, self._context
    )
  File "/usr/src/homeassistant/homeassistant/components/device_automation/action.py", line 72, in async_call_action_from_config
    await platform.async_call_action_from_config(hass, config, variables, context)
  File "/usr/src/homeassistant/homeassistant/components/select/device_action.py", line 121, in async_call_action_from_config
    await hass.services.async_call(
    ...<5 lines>...
    )
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solaredge_modbus_multi/select.py", line 291, in async_select_option
    await self._platform.write_registers(address=57357, payload=new_mode)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 1431, in write_registers
    await self.hub.write_registers(self.inverter_unit_id, address, payload)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 560, in write_registers
    result = await self._client.write_registers(
                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        self._wr_address, slave=self._wr_unit, values=self._wr_payload
[... 141 lines cut ...]
TypeError: object of type 'int' has no len()

### Debug logs

```shell

```

### Home Assistant Version

2025.02 and 2025.1 

### solaredge-modbus-multi Version

3.0.6 and 3.0.5 and 3.0.4

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-02-06:
Upload your diagnostic file.

sebwar2004, 2025-02-06:
> Upload your diagnostic file.

I've never done this before - let me know if this is the correct way:

Logger: homeassistant.components.websocket_api.http.connection
Quelle: components/websocket_api/commands.py:245
Integration: Home Assistant WebSocket API ([Dokumentation](https://www.home-assistant.io/integrations/websocket_api), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
Erstmals aufgetreten: 16:42:59 (1 Vorkommnisse)
Zuletzt protokolliert: 16:42:59

[140712750485936] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solaredge_modbus_multi/select.py", line 145, in async_select_option
    await self._platform.write_registers(address=57348, payload=new_mode)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 1431, in write_registers
    await self.hub.write_registers(self.inverter_unit_id, address, payload)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 560, in write_registers
    result = await self._client.write_registers(
                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        self._wr_address, slave=self._wr_unit, values=self._wr_payload
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 514, in write_registers
    return self.execute(no_response_expected, pdu_reg.WriteMultipleRegistersRequest(address=address, registers=values,dev_id=slave))
                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/pdu.py", line 36, in __init__
    self.count: int = count or len(self.registers)
                               ~~~^^^^^^^^^^^^^^^^
TypeError: object of type 'int' has no len()

WillCodeForCats, 2025-02-06:
Those are log messages.

Click the "download diagnostics" button in the device (Device info section) and upload that file.

sebwar2004, 2025-02-06:
[config_entry-solaredge_modbus_multi-85559acc66775d25886668e1707c2b02-7.json](https://github.com/user-attachments/files/18692930/config_entry-solaredge_modbus_multi-85559acc66775d25886668e1707c2b02-7.json)

WillCodeForCats, 2025-02-06:
The error:

```
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 514, in write_registers
    return self.execute(no_response_expected, pdu_reg.WriteMultipleRegistersRequest(address=address, registers=values,dev_id=slave))
                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/pdu.py", line 36, in __init__
    self.count: int = count or len(self.registers)
                               ~~~^^^^^^^^^^^^^^^^
TypeError: object of type 'int' has no len()
```

Is actually from pymodbus, not actually from the integration, so I'll have to try to figure out what they changed.

WillCodeForCats, 2025-02-06:
This integration works with 3.8.3 except for the write thing, and I don't know what pymodbus changes to cause this because the author didn't document it.

This is their API changes file:

https://github.com/pymodbus-dev/pymodbus/blob/dev/API_changes.rst

And it doesn't say anything about writes being different between 3.7 and 3.8 and they do not provide any write examples from documentation, so now I have to start digging through code which takes time.

sebwar2004, 2025-02-06:
ChatGPT interprets from the error message, that there might be a change in "WriteMultipleRegistersRequest".

_Two key changes stand out:

Removal of zero_mode in ModbusSlaveContext
If solaredge_modbus_multi previously used zero_mode=True, the addressing behavior (e.g., offsets) might now be different when writing values.
Stricter type checking in WriteMultipleRegistersRequest
In pymodbus 3.7.0, it was possible to pass a single integer value to WriteMultipleRegistersRequest.
In pymodbus 3.8.0, the registers parameter must be a list.
The error occurs when registers is an int because len(self.registers) is called, which does not work for int.
Possible Fix for the Developer
If solaredge_modbus_multi is currently calling WriteMultipleRegistersRequest with a single int, the solution could be to wrap the value in a list:

registers = [values] if isinstance(values, int) else values

Then, the line

self.count: int = count or len(self.registers)

should no longer throw an error, as len(self.registers) will only be called on lists._

I asked where it got the info with "WriteMultipleRegistersRequest" and this is the reply (link to docs doesn't work):

_I deduce this from several factors:

The error message itself:
File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/pdu.py", line 36, in __init__
  self.count: int = count or len(self.registers)
                            ~~~^^^^^^^^^^^^^^^^
TypeError: object of type 'int' has no len()
→ len(self.registers) fails because self.registers is an int.
→ self.registers is populated by WriteMultipleRegistersRequest(address=address, registers=values, dev_id=slave).
→ This suggests that values is an int, whereas pymodbus 3.8.0 expects a list.
Changes in pymodbus:
→ pymodbus 3.8.0 introduced stricter type checking. Previously, single values were implicitly accepted. Now, an explicit list must be passed.
Documentation for WriteMultipleRegistersRequest:
→ The method expects registers: list[int].
→ Source: [pymodbus Documentation](https://pymodbus.readthedocs.io/en/latest/source/library/pdu.html)
Comparison with previous pymodbus versions:
→ Older versions allowed registers to be a single integer.
→ The current implementation calls len(self.registers), which fails if self.registers is an integer._

frank8the9tank, 2025-02-06:
my diagnostics file:

[config_entry-solaredge_modbus_multi-01JKDH323Z502S1QG92QS3GKPR.json](https://github.com/user-attachments/files/18694061/config_entry-solaredge_modbus_multi-01JKDH323Z502S1QG92QS3GKPR.json)

WillCodeForCats, 2025-02-06:
I will update it as quickly as possible, unfortunately it's not as simple as changing the payload to a list format for all write functions. On my development system it worked for some but not all of them.

WillCodeForCats, 2025-02-06:
Please try [Release v3.1.0-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.1.0-pre.1) to fix the write errors.

sebwar2004, 2025-02-07:
Works with v3.1.0-pre.1 👍 

What I don't get - why did it work also with v3.0.6, as long as the other integration (in my case Huawei) wasn't loaded?
I've seen that [frank8the9tank](https://github.com/frank8the9tank) also has another Modbus integration next to SolarEdge Modbus Multi.

WillCodeForCats, 2025-02-07:
Because of the conflicting version requirements for `pymodbus`. The solax or huawei custom integrations set a fixed pymodbus version of 3.8.3 which conflicted with my supported version range of 3.6.6-3.7.4, and only one version of pymodbus could load at a time.

Schlichi83, 2025-02-07:
I have the same problem. How can I update to toe 3.1.0-pre1 version?

sebwar2004, 2025-02-07:
> I have the same problem. How can I update to the 3.1.0-pre1 version?

Search for the Integration in HACS and choose "Download again" (I only have the German wording) via the three dots and them you can choose a different version.

Schlichi83, 2025-02-07:
Thank you - that worked!

glenmclarke, 2025-02-07:
I think I have a variation of the same issue - but with the 'Limit Control Mode' on an SE10000H-AUSNBBX14 inverter.

After updating to Core 2025.2.0, the Limit Control Mode selector gave the same 'Failed to perform the action select/select_option. object of type int has no len()' as described above when changing options.

After updating to your release v3.1.0_pre.1, I can now select other options, but now I cannot go back to the 'Disabled' option. If I select 'Disabled', I receive a new error - see below.

[image]

I have attached the diagnostics file:

[config_entry-solaredge_modbus_multi-d30afef8706830a3b5c43a5600139846.json](https://github.com/user-attachments/files/18707086/config_entry-solaredge_modbus_multi-d30afef8706830a3b5c43a5600139846.json)

BTW - thanks for the integration, it has been incredibly useful.

WillCodeForCats, 2025-02-07:
> After updating to Core 2025.2.0, the Limit Control Mode selector gave the same 'Failed to perform the action select/select_option. object of type int has no len()' as described above when changing options.

Fixed in [Release v3.1.0-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.1.0-pre.2)

glenmclarke, 2025-02-08:
Confirmed, that fixed the problem - Thank you.

frank8the9tank, 2025-02-08:
For me this looks fixed for now, i will check the other integrations that use pymodbus. what what happens when i change there dependency/version. Didn't known that one integration could force this onto a other integration.
So i'm i correct to say that when a other integration has a fixed pymodbus version that is locks it for all other integrations?

WillCodeForCats, 2025-02-09:
Yes, each integration that uses locked versions will try to use that version of pymodbus, but only one version can be installed at a time by Home Assistant. The HA team likes to use locked versions, so most custom integrations follow that practice too.

I try not to use locked versions because I don't want to make a release every time the version change, so for these changes I've set 3.8.3 or higher which means it will work with 3.8.3 or the new 3.8.4 which was just released if HA starts using that version (or higher) in the next release. So I watch their API changes file and look for breaking changes.

closed 2025-02-10 by WillCodeForCats