WillCodeForCats/solaredge-modbus-multi #764: 3.1.0: pymodbus.exceptions.ModbusException: Modbus Error: Illegal size (3) of register array, cannot convert!

rud-ale, 2025-02-25:
### Describe the bug

After docker container fresh start:
pymodbus.exceptions.ModbusException: Modbus Error: Illegal size (3) of register array, cannot convert!

This happens all the time, integration can not be started successfully.
Only downgrade using backup to 3.0.6 helps.

### Expected behavior

New version 3.1.0 should be startable und run smoothly as  version 3.0.6 do.

### Diagnostic File

[solaredge-modbus-multi.protokoll.log](https://github.com/user-attachments/files/18970186/solaredge-modbus-multi.protokoll.log)
[home-assistant_solaredge_modbus_multi_2025-02-25T17-52-07.069Z.log](https://github.com/user-attachments/files/18970187/home-assistant_solaredge_modbus_multi_2025-02-25T17-52-07.069Z.log)

### Debug logs

```shell

```

### Home Assistant Version

Core 2025.2.5

### solaredge-modbus-multi Version

3.1.0

### Installation Type

Container

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-02-25:
I will need debug logs of it trying to start showing the error.

rud-ale, 2025-02-25:
[home-assistant_solaredge_modbus_multi_2025-02-25T17-52-07.069Z.log](https://github.com/user-attachments/files/18970187/home-assistant_solaredge_modbus_multi_2025-02-25T17-52-07.069Z.log)

Thought this is the debug - log ? 
If not, i will see where i can find it ....

WillCodeForCats, 2025-02-25:
Sorry, I meant the diagnostic file too. Click download diagnostics from the menu.

[image]

rud-ale, 2025-02-25:
Ah, okay. Sorry, my first issue ... ;)
Just give me some seconds. For that i have to reinstall the 3.1.0 ...

rud-ale, 2025-02-25:
Well, sorry, it seems to be not possible in this state...
[image]

rud-ale, 2025-02-25:
I can activate the debug protocol.
Reload the integration.
Deacticate the debug protocol.

[home-assistant_solaredge_modbus_multi_2025-02-25T18-30-38.009Z.log](https://github.com/user-attachments/files/18970569/home-assistant_solaredge_modbus_multi_2025-02-25T18-30-38.009Z.log)

But it seems like it's the same i uploaded earlier.

rud-ale, 2025-02-25:
Some additional info:
System is trying to restart, but will not work.

I have seen this 2 hints:
Einrichtungsfehler, wird erneut versucht: Modbus Error: Illegal size (3) of register array, cannot convert!
Einrichtungsfehler, wird erneut versucht: Exception Response(131, 3, None)

Perhaps it helps. 
For now i will downgrade using backup to get it back working...

In any case, thank you in advance for taking a look...!

WillCodeForCats, 2025-02-25:
Error `Exception Response(131, 3, None)` means a feature on the inverter isn't supported, that one is normal to see on startup.

I am working on trying to duplicate the illegal size error.

rud-ale, 2025-02-25:
I restored my backup - 3.0.6 is back and running.

In any case, thank you very much for that and of course for the great integration - it normally works without any problems.

WillCodeForCats, 2025-02-25:
Try removing `solaredge_modbus` integration, it has a conflicting pymodbus version requirement.

WillCodeForCats, 2025-02-25:
My release v3.1.0 was updated for Home Assistant 2025.2 and pymodbus 3.8.x. Uninstall or delete the `solaredge_modbus` integration from HA and restart should fix the problem.

The `solaredge_modbus` integration requires an older version of pymodbus. When it starts after my integration it forces the old version of pymobus to load, which will break any integrations or HA core that was updated to work with newer versions. Release v3.0.6 will work with the old versions of pymodbus.

In the debug logs look for this line:

`2025-02-25 18:52:02.695 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] pymodbus version 3.7.4
`

This shows the old version was loaded. It needs to be version 3.8.3 or higher.

WillCodeForCats, 2025-02-25:
My integration can do commands too. You will need to turn them on under Configure and Power Control Options:

[image]

This is what my testing dashboard looks like:

[image]

rud-ale, 2025-02-26:
Hey WillCodeForCats.

Thank you very much for your help.

I hadn't used the "Advanced Power Control" because I couldn't estimate exactly what was activated. Reading and understanding the manual would have helped.

I have now activated this option, deactivated all entities that I don't need, changed all references from solaredge_modbus_... to solaredge_modubus_i1_... and updated to 3.1.0.

I can't find any more problems.

THANK YOU!

closed 2025-02-26 by rud-ale