WillCodeForCats/solaredge-modbus-multi #768: Add support for individual Phase Power A-C (Modbus 40207 - 40209)

Piranha1511, 2025-02-27:
**Describe the feature**
It would be cool to have the ability to get also the individual AC Power per phase (L1-L3) to pass to Victron VenusOS. This is required for the new virtual pv inverter emulation in VenusOS.

**Screenshots**
If applicable, add screenshots or edits of the proposed feature that help describe it.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2025.2.05
 - solaredge-modbus-multi Version: 3.1.0

**Additional context**
Add any other context about the request here.

WillCodeForCats, 2025-02-27:
All possible data is already provided.

You can always add meters to your system if you don't have any.

Piranha1511, 2025-02-27:
Why should I add additional physical meters if the data is provided via modbus. According to the spec of SolarEdge there are additional modus registers, which are not supported or exposed through this integration. I already can read those registers via modbus directly. 

[image]

WillCodeForCats, 2025-02-27:
Then upload your diagnostic file so I can see what you're doing with the integration.

I guarantee all documented registers are supported.

WillCodeForCats, 2025-02-27:
Show me your manual reads, too.

Piranha1511, 2025-02-27:
Do you know that file? There are a lot of more modbus parameters which are not yet supported. 

[Modbus-Register SE.xlsx](https://github.com/user-attachments/files/19014151/Modbus-Register.SE.xlsx)

WillCodeForCats, 2025-02-27:
I'm trying to help, and you need to answer my questions. 

Or just look at the code in hub.py file and docs folder: the proof is there.

WillCodeForCats, 2025-02-27:
Turn on debug logging and reload the integration to show the startup sequence as well.

Piranha1511, 2025-02-27:
I think we talk about something different. I don't have an issue with your integration which needs to be analyzed. I have a feature request, if you can support additional parameters which are available through Modbus TCP. 
You can also check the Modbus Sunspec here: https://knowledge-center.solaredge.com/sites/kc/files/sunspec-implementation-technical-note.pdf

On page 22 first 4 lines are the modbus registers I'm talking about. I'm reading them already via Loxone Modbus TCP. But would be cool to have them directly implemented in your integration

WillCodeForCats, 2025-02-27:
If you want my help stop sending me links to files I do not care about, answer my questions, and upload your debug logs showing integration startup and your diagnostic file.

You think you are asking for something new BUT YOU ARE NOT. My integration is arguably the most complete one available for SolarEdge. I have worked very hard for several years to make a complete integration. You have a problem nobody else using the integration has.

I don't know what you're doing but you won't help me figure out what because you keep sending me things I don't want to see. So I can't help you.

Piranha1511, 2025-02-27:
[home-assistant_solaredge_modbus_multi_2025-02-27T20-58-39.991Z.log](https://github.com/user-attachments/files/19016246/home-assistant_solaredge_modbus_multi_2025-02-27T20-58-39.991Z.log)

[config_entry-solaredge_modbus_multi-01JMVY3743MY66NXM1TMYB8HZ5 (1).json](https://github.com/user-attachments/files/19016249/config_entry-solaredge_modbus_multi-01JMVY3743MY66NXM1TMYB8HZ5.1.json)

WillCodeForCats, 2025-02-27:
Okay so these are the important sections of the startup sequence where it looks to see if there is an attached meter to identify:

Meter slot 1:
```
2025-02-27 21:58:30.991 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for meter I125M1
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 67 67
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: C_SunSpec_DID 0x2bd <class 'int'>
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: C_SunSpec_Length 0x99 <class 'int'>
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: C_Device_address 0x8000 <class 'int'>
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: C_Manufacturer 	' <class 'str'>
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: C_Model h <class 'str'>
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: C_Option  <class 'str'>
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: C_Version 	' <class 'str'>
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: C_SerialNumber  <class 'str'>
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M1: Meter 1 ident incorrect or not installed.
```

Meter slot 2:
```
2025-02-27 21:58:31.014 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for meter I125M2
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 67 67
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: C_SunSpec_DID 0xffff <class 'int'>
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: C_SunSpec_Length 0x0 <class 'int'>
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: C_Device_address 0xffff <class 'int'>
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: C_Manufacturer  <class 'str'>
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: C_Model  <class 'str'>
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: C_Option  <class 'str'>
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: C_Version  <class 'str'>
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: C_SerialNumber A <class 'str'>
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M2: Meter 2 ident incorrect or not installed.
```

Meter slot 3:
```
2025-02-27 21:58:31.043 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for meter I125M3
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 67 67
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: C_SunSpec_DID 0x0 <class 'int'>
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: C_SunSpec_Length 0x0 <class 'int'>
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: C_Device_address 0x0 <class 'int'>
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: C_Manufacturer / <class 'str'>
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: C_Model  <class 'str'>
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: C_Option  <class 'str'>
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: C_Version  <class 'str'>
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: C_SerialNumber i <class 'str'>
2025-02-27 21:58:31.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I125M3: Meter 3 ident incorrect or not installed.
```

All of the responses to `C_SunSpec_DID` and `C_SunSpec_Length` are bad (ident incorrect or not installed), so it will not poll meter registers unless it identifies as a meter per specifications.

[image]

`C_SunSpec_DID` must have a response of `0x1` and `C_SunSpec_Length` must be `0x41`.

And it is returning garbage in the other meter common block registers for model, manufacture, version, option, serial number.

This is what a good meter slot response looks like:

```
2025-02-27 13:23:18.970 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for meter I1M1
2025-02-27 13:23:19.021 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Registers received requested : 67 67
2025-02-27 13:23:19.022 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_SunSpec_DID 0x1 <class 'int'>
2025-02-27 13:23:19.022 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1M1: C_SunSpec_Length 0x41 <class 'int'>
[... cut ...]
This is why I need to see debug logs and diagnostic files, because it's clear the problem is not the integration. The problem is the device it's communicating with does not identify a meter in any of the three meter slots.

Without that first step checking C_SunSpec_DID` and `C_SunSpec_Length` it will not create a meter device in Home Assistant, and without the meter device it can't read the meter registers 40188 through 40294. The integration will not read random registers unless all of the supporting conditions for those registers pass.

Piranha1511, 2025-02-27:
I don't have a physical meter directly attached to the SolarEdge inverters, since I have a Sonnenbatterie in parallel controlling the inverter. But all I can say is that I can easily get access to Modus register via Loxone Modbus TCP. 

[image]
[image]

WillCodeForCats, 2025-02-27:
That won't work, you need to add all of the other registers the integration is looking for to identify a meter.

Piranha1511, 2025-02-27:
Maybe that is a way to look at it. There are installations where the control of the inverter is managed externally through another device or system (like in my case a Sonnenbatterie, containing the SmartMeter). Same with some of Victron installations, I'm aware about. 

Where does this meter requirement come from and is it really necessary? Because the above Loxone configuration is only opening a Modbus TCP connection to the SolarEdge inverter and requesting the data for the relevant modbus register. There is no requirement for an attached meter.

[image]

WillCodeForCats, 2025-02-27:
Yes it's necessary for this integration.

You linked it, it's also all in the [doc/](https://github.com/WillCodeForCats/solaredge-modbus-multi/tree/main/doc) folder.

If some other software like Loxone chooses only to pick single registers from the inverter that is their design choice. My design choice is to implement the complete three meter slots as described by SolarEdge and perform detection routines to make it easy for non-technical users to install this integration without having to know what modbus registers are.

WillCodeForCats, 2025-02-27:
You can make it work, just fix the virtual one to respond to registers 40121 through 40187 and give them types/values from the SolarEdge docs.

You will also need to it to respond to 40188 with one of the values `[201, 202, 203, 204]` which determines the type of meter, and 40189 should return `0x69`.

MMelzig, 2025-02-28:
Just one remark:

Because the sunspec provides only the overall power values I_AC_POWER resp. I_DC_POWER and no individual power values per phase I had a discussion with Solaredge some days ago concerning the determination of individual power of the inverter per phase. 

The answer was the solaredge inverter **distributes its power equally** to the phases. With other words, you have 1/3 I_AC_POWER resp. I_DC_POWER per phase. I own a SE10K. 

I have a meter from Solaredge to measure values to the grid and it can be read out with this add-on. And this provides the value M_AC_POWER_A to ..._C from/to the grid. But the way the meter is used in my system, it does NOT measure the individual phase power of the inverter.

WillCodeForCats, 2025-02-28:
Another suggestion if you're only trying to read 5 registers from something, then you don't need to use an integration that's designed to be complete. Just use the [core modbus integration](https://www.home-assistant.io/integrations/modbus/).

Otherwise, to use this integration you need to pass through a complete set of modbus registers from your virtual inverter.

closed 2025-03-01 by WillCodeForCats