WillCodeForCats/solaredge-modbus-multi #786: Only one of my two batteries are discovered

BostonDanPhD, 2025-03-15:
### Describe the bug

I have a solaredge system with two inverters (se6000 and se7600), and two lg chem batteries (RESU10H and RESU16H), with one battery connected to a separate inverter. In solaredge-mosbus-multi, I configure the system to discover batteries, but only the first battery (RESU10H), which is connected to the leader inverter, is discovered.

### Expected behavior

Both batteries are discovered.

### Diagnostic File

[config_entry-solaredge_modbus_multi-01JPARRHSJ1HM9PAC9A9F716XB-1.json](https://github.com/user-attachments/files/19263009/config_entry-solaredge_modbus_multi-01JPARRHSJ1HM9PAC9A9F716XB-1.json)

### Debug logs

```shell

```

### Home Assistant Version

25.3.3

### solaredge-modbus-multi Version

3.1.0

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-03-15:
The detection cycle is the same loop for all inverters. You can turn on debug logging and reload the integration to show the startup cycle, but if an inverter doesn't proxy the battery there's nothing I can do about that. You may have to try contacting SolarEdge support, see known issues in the wiki for more discussions.

Also I see you have `solaredge_modbus` installed: you really need to make sure it NEVER tries to load becasue if it does, it will load an old version of pymodbus which will cause problems with integrations that are updated for newer pymobus and Home Assistant. I can only recommend removing that completely.

BostonDanPhD, 2025-03-15:
Thank you for the quick response. I will contact solaredge. I did not see the known issue on this in the wiki, so I will try to be more diligent in my investigation.

I only have solaredge_modbus to try to resolve this issue, and I was disabling either solaredge_modbus while using solaredge_modbus_multi or vice versa, as I am well aware that only one connection to modbus is permitted at a time.

Again, I really appreciate the information and support.

WillCodeForCats, 2025-03-15:
It's not just the connection, but if HA loads `solaredge_modbus` it will cause problems internally with the outdated pymodbus version that integration requires.

When that happens you will see it log random errors like reported n this issue: https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/764.

BostonDanPhD, 2025-03-15:
Ah. Understood. I have removed the component.

Again, appreciate the information.

WillCodeForCats, 2025-03-15:
Turn on debug logging, reload the integration, and then upload the log so I can see the startup detection to confirm.

BostonDanPhD, 2025-03-15:
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.3.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.2",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "6.6.62-haos-raspi",
    "supervisor": "2025.03.2",
    "host_os": "Home Assistant OS 14.2",
    "docker_version": "27.2.0",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "sunspec": {
      "documentation": "https://github.com/cjne/ha-sunspec",
      "version": "0.0.26",
      "requirements": [
        "pysunspec2==1.1.5"
      ]
    },
    "ember_mug": {
      "documentation": "https://github.com/sopelj/hass_ember_mug",
      "version": "1.3.0",
      "requirements": [
        "python-ember-mug==1.1.0"
      ]
    },
    "solaredge_modbus_multi": {
      "documentation": "https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki",
      "version": "3.1.0",
      "requirements": [
        "pymodbus>=3.8.3"
      ]
    },
    "xfinity": {
      "documentation": "https://github.com/robert-alfaro/xfinity-usage",
      "version": "0.1.0",
      "requirements": []
    },
    "attribute_as_sensor": {
      "documentation": "https://github.com/gjohansson-ST/attribute_as_sensor/blob/master/readme.md",
      "version": "1.6",
      "requirements": []
    },
    "grocy": {
      "documentation": "https://github.com/custom-components/grocy",
      "version": "2025.2.2",
      "requirements": [
        "pygrocy2==2.4.0"
      ]
    },
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "teamtracker": {
      "documentation": "https://github.com/vasqued2/ha-teamtracker",
      "version": "0.1",
      "requirements": [
        "arrow",
[... 564 lines cut ...]
        "B_SOE": "0x422ffada",
        "B_Export_Energy_WH": "0x559",
        "B_Import_Energy_WH": "0x519",
        "B_Status": "0x3",
        "B_Status_Vendor": "0x3",
        "B_Event_Log1": "0x0",
        "B_Event_Log2": "0x0",
        "B_Event_Log3": "0x0",
        "B_Event_Log4": "0x0",
        "B_Event_Log5": "0x0",
        "B_Event_Log6": "0x0",
        "B_Event_Log7": "0x0",
        "B_Event_Log8": "0x0",
        "B_Event_Log_Vendor1": "0x0",
        "B_Event_Log_Vendor2": "0x0",
        "B_Event_Log_Vendor3": "0x0",
        "B_Event_Log_Vendor4": "0x0",
        "B_Event_Log_Vendor5": "0x0",
        "B_Event_Log_Vendor6": "0x0",
        "B_Event_Log_Vendor7": "0x0",
        "B_Event_Log_Vendor8": "0x0"
      }
    }
  }
}

WillCodeForCats, 2025-03-15:
[image]

Turn on debug logging, reload the integration, then turn off. When debug logging is turned off it will download a log file.

BostonDanPhD, 2025-03-15:
[error_log.txt](https://github.com/user-attachments/files/19263737/error_log.txt)

WillCodeForCats, 2025-03-15:
Yes, I can confirm the inverter is returning unsupported address for the battery locations on inverter unit 2:

```
2025-03-15 11:42:25.593 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for battery I2B1
2025-03-15 11:42:25.825 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-15 11:42:25.826 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Unit 2 Read IllegalAddress: ExceptionResponse(dev_id=2, transaction_id=14, address=0, count=0, bits=[], registers=[], status=1)
2025-03-15 11:42:25.826 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B1: Battery 1 unsupported address
```

```
2025-03-15 11:42:25.826 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for battery I2B2
2025-03-15 11:42:26.078 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-15 11:42:26.081 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Unit 2 Read IllegalAddress: ExceptionResponse(dev_id=2, transaction_id=15, address=0, count=0, bits=[], registers=[], status=1)
2025-03-15 11:42:26.083 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B2: Battery 2 unsupported address
```

```
2025-03-15 11:42:26.083 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Looking for battery I2B3
2025-03-15 11:42:26.474 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-15 11:42:26.482 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Unit 2 Read IllegalAddress: ExceptionResponse(dev_id=2, transaction_id=16, address=0, count=0, bits=[], registers=[], status=1)
2025-03-15 11:42:26.483 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I2B3: Battery 3 unsupported address
```