WillCodeForCats/solaredge-modbus-multi #790: SolarEdge Meter energy export/import entities are not updated

a-lenkov, 2025-03-19:
### Describe the bug

All SolarEdge Meter

[image]
[image]

### Expected behavior

 energy export/import entities have to be updated during the time.

### Diagnostic File

[home-assistant_solaredge_modbus_multi_2025-03-19T16-36-46.722Z.log](https://github.com/user-attachments/files/19347936/home-assistant_solaredge_modbus_multi_2025-03-19T16-36-46.722Z.log)

### Debug logs

```shell

```

### Home Assistant Version

2025.3.3

### solaredge-modbus-multi Version

3.1.1

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-03-19:
The number data doesn't change from the inverter so this is correct display.

`I1M1: AC_Energy_WH_Exported_A 0xb71486f <class 'int'>`

This value is not changing so this is not an integration bug.

closed 2025-03-19 by WillCodeForCats

WillCodeForCats, 2025-03-19:
You can try power cycling your inverter:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues