WillCodeForCats/solaredge-modbus-multi #792: Battery measurement unstable

erich-maier, 2025-03-20:
### Describe the bug

Hello all,

i have the situation that my energy board did not show me the battery in or battery out measurements. Sometime it is shown, sometimes not. 

The battery works absolutely correct. All other settings are also fine. I have found something similar in an other project. https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/281 could it be the same?

I use:
Core:  2025.3.3
Supervisor: 2025.03.3
Operating System: 15.0
Frontend: 20250306.0

### Expected behavior

Have look into the device of SolarEdge batteries

### Diagnostic File

[image]

### Debug logs

```shell

```

### Home Assistant Version

2025.3.3

### solaredge-modbus-multi Version

3.1.1

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

erich-maier, 2025-03-20:
yes I found that messages in the log files. But that did not match with timing of unavailability of sensors ...

WillCodeForCats, 2025-03-20:
That's why they go into unavailable state, because watt-hour sensors should never go backwards.

You can choose how you want to handle that condition with some options:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration#battery-options

You can turn on Allow Battery Energy to Reset or set Update Cycles to Reset Battery Energy to a value greater than zero.

I don't personally have batteries so I can't recommend the best settings for this.

There is more information under Battery Energy in Known Issues:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues#battery-energy

But this is not an integration bug.