WillCodeForCats/solaredge-modbus-multi #794: 8 !=40

w1Ngx, 2025-03-21:
### Describe the bug

any idea what this error is about? 

Logger: custom_components.solaredge_modbus_multi.hub
Source: custom_components/solaredge_modbus_multi/hub.py:539
integration: SolarEdge Modbus Multi ([documentation](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki), [issues](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues))
First occurred: 7:19:21 PM (4 occurrences)
Last logged: 9:37:07 PM

Registers received != requested : 8 != 40
Registers received != requested : 40 != 2

### Expected behavior

no error in log

### Diagnostic File

[config_entry-solaredge_modbus_multi-01JJYW0NTXBRGJS1XVK6JRCKX4.json](https://github.com/user-attachments/files/19397944/config_entry-solaredge_modbus_multi-01JJYW0NTXBRGJS1XVK6JRCKX4.json)

### Debug logs

```shell

```

### Home Assistant Version

2025.3.3

### solaredge-modbus-multi Version

3.1.1

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-03-22:
It means it requested X number of registers but didn't get X back.

If you are running a proxy then the proxy had an error.

w1Ngx, 2025-03-24:
Hm, i am running nginx, can this be connected?

WillCodeForCats, 2025-03-25:
Sure, any proxy can cause that if it mixed up responses to concurrent requests.

closed 2025-03-26 by WillCodeForCats