WillCodeForCats/solaredge-modbus-multi #798: Wattnode entities visible, but no values

buchulator, 2025-03-23:
### Describe the bug

Hi,

I managed to connect my old Solaredge SE-7K via Modbus TCP to Homeassistant.
All SE-7K values are readable. The values delivered by the Wattnode meter however, which is connected to the SE7K via RS485, are all not readable. That is, entities are all there, but without values.
Any idea ?

Regards

### Expected behavior

There should be values on the Wattnode meter like AC current, AC Power etc.

### Screenshots

[image]
[image]

### Diagnostic File

[config_entry-solaredge_modbus_multi-01JQ1Q1TCPKFQHVEVVXW3KMY6Z (1).json](https://github.com/user-attachments/files/19409835/config_entry-solaredge_modbus_multi-01JQ1Q1TCPKFQHVEVVXW3KMY6Z.1.json)

### Debug logs

```shell

```

### Home Assistant Version

2025.3.4

### solaredge-modbus-multi Version

3.1.1

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-03-24:
All of your meter data values in the diagnostic file are `0x0` or `-0x8000`.

While these are not useful values they are what the inverter is returning, and not a bug in the integration.

You could try power cycling the system.

There is also a non-zero meter events value, check the meter events sensor under diagnostics.

buchulator, 2025-03-24:
Power cycling did not help. In the Solaredge App (cloud based) I have alle meter readings including export, import and so on with no errors. So meter should be ok, and inverter probably too.. maybe it is a firmware bug regarding modbus data inside SE7K inverter ?

closed 2025-03-25 by WillCodeForCats