WillCodeForCats/solaredge-modbus-multi #810: Solaredge modbus sometimes unable to connect

FrankTub, 2025-04-20:
### Describe the bug

In my setup the data is refreshed every 5 seconds. However, I quite regularly see all sorts of errors show up. 

The different types of errors I see are:

```log
Logger: pymodbus.logging
Source: custom_components/solaredge_modbus_multi/hub.py:492
integration: SolarEdge Modbus Multi ([documentation](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki), [issues](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues))
First occurred: April 19, 2025 at 6:23:24 PM (315 occurrences)
Last logged: 3:23:32 PM

Failed to connect
Failed to connect [Errno 111] Connect call failed ('192.168.68.107', 1502)
Failed to connect [Errno 113] Connect call failed ('192.168.68.107', 1502)
```

and

```log
Logger: pymodbus.logging
Source: custom_components/solaredge_modbus_multi/hub.py:523
integration: SolarEdge Modbus Multi ([documentation](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki), [issues](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues))
First occurred: April 19, 2025 at 11:27:55 PM (130 occurrences)
Last logged: 3:11:54 PM

Cancel send, because not connected!
```

and 

```log
Logger: custom_components.solaredge_modbus_multi.hub
Source: custom_components/solaredge_modbus_multi/hub.py:552
integration: SolarEdge Modbus Multi ([documentation](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki), [issues](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues))
First occurred: April 19, 2025 at 11:27:51 PM (62 occurrences)
Last logged: 3:11:47 PM

Registers received != requested : 40 != 2
Registers received != requested : 8 != 40
```

Not really sure if this is a bug or expected behaviour given my setup.

I think I followed along with the [configuration](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration), that all seems fine. The thing I'm not sure about is that I only have a single inverter, however my setup looks like:

[image]

[image]

The

### Expected behavior

No errors and always able to connect

### Diagnostic File

[config_entry-solaredge_modbus_multi-01JKKFW47NRP3Z8ZB2E9GDVH2B.json](https://github.com/user-attachments/files/19825485/config_entry-solaredge_modbus_multi-01JKKFW47NRP3Z8ZB2E9GDVH2B.json)

### Debug logs

```shell

```

### Home Assistant Version

2025.4.3

### solaredge-modbus-multi Version

3.1.2

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-04-20:
Are you using any type of proxy? The `!=` errors means we asked for X number of registers but the response did not match what was asked for. If we ask for X we should always receive X, not different number Y. If you are using a proxy that means it's mixing up responses.

You can set both RS485 ports to "None" if nothing is connected to them.

FrankTub, 2025-04-20:
My solaredge and home assistant installation are directly on my wifi network, so I'm not using a proxy. This is my configuration of the integration:

[image]

> You can set both RS485 ports to "None" if nothing is connected to them.

Not really sure if I'm following what you mean here?

WillCodeForCats, 2025-04-20:
The RS485 ports have a "None" option. You sent screenshots of it, go back there and change from Follower to None if you actually have nothing connected to those ports.

FrankTub, 2025-04-20:
I have tried to setup my Solaredge inverter to use a network cable that is attached to my deco. For some reason I'm unable to get it to work and use the ethernet cable. Any advice on that? 

And how do I know if I have something attached to one of those RS485 ports?

locussss, 2025-04-22:
Hi!

Don't know if this is related but every day around 22.30 - 22.45 I get these:

```
2025-04-19 22:43:46.295 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-19 22:43:50.201 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-19 22:43:55.705 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-19 22:44:06.009 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-19 22:44:27.715 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.2.215', 1502)
2025-04-19 22:44:27.817 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Error fetching SolarEdge Coordinator data: Modbus/TCP connect to 192.168.2.215:1502 failed.
2025-04-20 22:39:30.295 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-20 22:39:34.199 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-20 22:39:38.756 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.2.215', 1502)
2025-04-20 22:39:46.071 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.2.215', 1502)
``` 

This is with:

[image]

roma777roma, 2025-04-22:
same problem:

Questo errore ha avuto origine da un'integrazione personalizzata.

Registratore: pymodbus.logging
Fonte: custom_components/solaredge_modbus_multi/hub.py:492
Integrazione: SolarEdge Modbus Multi (documentazione, problemi)
Prima occorrenza: 22 aprile 2025 alle ore 20:01:33 (2295 occorrenze)
Ultimo accesso: 00:50:41

Failed to connect [Errno 111] Connect call failed ('192.168.0.42', 1502)
Failed to connect
Failed to connect [Errno 113] Connect call failed ('192.168.0.42', 1502)

after update 2025.4.3, 2025.4.1 work fine.

scholzt, 2025-04-29:
Same Problem.

After Upgrading to latest core (2025.4.3) I now receive connection Problems:
Failed to connect [Errno 111] Connect call failed

WillCodeForCats, 2025-04-29:
That's correct, it's impossible to completely eliminate connection failures when the inverter resets/reboots or updates or whatever it's actually doing. Faster polling times will see this more often. I personally use a 30 second polling time.

When the inverter is sleeping and does its reset it will not accept modbus connections. It won't do the reset during solar production, so it should only happen at night.

FrankTub, 2025-04-30:
These errors indeed often happen late in the evening. I guess the errors during the day are only happening during the day when wifi is unstable or something and haven't had any of that lately. @WillCodeForCats I checked the wiki but there is no documentation on how and where to attach the ethernet cable and change the settings on the inverter right? I have tried to attach an ethernet cable however in the settings when trying to switch to cabled internet it does not detect it. Any advice on this?

Furthermore I was wondering if something in my setup is wrong that sometimes I get the error:

```
Registers received != requested : 40 != 2
Registers received != requested : 8 != 40
```

I have a single inverter and no battery so not sure why this is popping up.