WillCodeForCats/solaredge-modbus-multi #811: Problem with modbus connection after inverter update to CPU 4.23.22

danymexi, 2025-04-22:
### Describe the bug

After performing an update to my inverter I can't connect anymore and I keep getting this error message:

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solaredge_modbus_multi/hub.py:492
integration: SolarEdge Modbus Multi (documentation, issues)
First occurred: 21:12:00 (16 occurrences)
Last logged: 21:23:40

Failed to connect [Errno 111] Connect call failed ('192.168.178.235', 1502)

### Expected behavior

Everything is set up correctly. I've been using the integration for years now... started to have problems today after performing the Inverter CPU update.  

### Diagnostic File

2025-04-22 20:14:44.288 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solaredge_modbus_multi which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-22 20:15:50.305 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ (states('sensor.solaredge_b1_state_of_energy') | float + states('sensor.solaredge_b2_state_of_energy') | float + states('sensor.solaredge_b3_state_of_energy') | float) / 3 }}) renders=2>
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.solaredge_b1_state_of_energy') | float + states('sensor.solaredge_b2_state_of_energy') | float + states('sensor.solaredge_b3_state_of_energy') | float) / 3 }}' but no default was specified
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.solaredge_b1_state_of_energy') | float + states('sensor.solaredge_b2_state_of_energy') | float + states('sensor.solaredge_b3_state_of_energy') | float) / 3 }}' but no default was specified
2025-04-22 20:15:50.308 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (states('sensor.solaredge_b1_state_of_energy') | float + states('sensor.solaredge_b2_state_of_energy') | float + states('sensor.solaredge_b3_state_of_energy') | float) / 3 }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.solaredge_b1_state_of_energy') | float + states('sensor.solaredge_b2_state_of_energy') | float + states('sensor.solaredge_b3_state_of_energy') | float) / 3 }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.batteria_media'
2025-04-22 20:15:50.649 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ states('sensor.solaredge_m1_ac_power') | float | abs }}) renders=2>
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states('sensor.solaredge_m1_ac_power') | float | abs }}' but no default was specified
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states('sensor.solaredge_m1_ac_power') | float | abs }}' but no default was specified
2025-04-22 20:15:50.650 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ states('sensor.solaredge_m1_ac_power') | float | abs }}' but no default was specified') while processing template 'Template<template=({{ states('sensor.solaredge_m1_ac_power') | float | abs }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.solaredge_ac_power_positivo'
2025-04-22 20:15:51.346 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_lifetime_energy as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.346 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_current_power as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_sensors as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_batteries as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_power_consumption as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_solar_power as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_grid_power as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_storage_power as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_imported_energy as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_production_energy as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_consumption_energy as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_selfconsumption_energy as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_exported_energy as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.347 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_test_storage_level as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.357 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_b1_status_text as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.360 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_i1_ac_current as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.360 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_i1_ac_voltage_ab as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.360 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_i1_ac_power as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.360 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_i1_ac_frequency as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.360 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_i1_ac_energy as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.360 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_i1_dc_current as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.360 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_i1_dc_voltage as this would exceed the 150 device limit. Consider using the filter option
2025-04-22 20:15:51.360 WARNING (MainThread) [homeassistant.components.homekit] Cannot add sensor.solaredge_i1_dc_power as this would exceed the 150 device limit. Consider using the filter option
[... cut ...]

2025.4.3

### solaredge-modbus-multi Version

v3.1.2

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-04-22:
Try turning off Auto Detect Additional Entities.

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration#integration-configuration-options

WillCodeForCats, 2025-04-22:
Also check that modbus is still enabled in the inverter.

But the integration is not inverter version specific, so this would be an inverter problem related to updating its firmware, not an integration issue.

danymexi, 2025-04-22:
It's back working now... I had to physically shut down the entire system and remove the auto-detect feature for additional entities. Thanks a lot

thomas-ruw, 2025-04-23:
Hi, 

I have the same problem. The inverter updated to 4.23.22. After that I had no modbus connection.

In the inverter settings I disabled and enabled TCP modbus. After that everything was working without rebooting the inverter.
But today in the morning again no connection. 

I'll try to disable the auto detect options, later today when I'm back at home.

bartolomeo902, 2025-04-23:
I have the same issue. My installer technician updated the firware to 4.23.22 and the same problem occured to me as well.
It works fine for few hours after a complete FV system power cycle, then it breaks again. ModBus configuration is unchanged and I followed all the suggestions.
@thomas-ruw did you have any luck disabling the auto detect options?

[image]

I assume I need both inverter and battery auto-detect funcions enabled in order for the integration to detect the formers right?

goofv, 2025-04-23:
Same problem here. after a system power cycle i t works till the next morning, and then it cant connect anymore

bartolomeo902, 2025-04-23:
@danymexi same here. However, stopped again the next morning

AMichaelSuess, 2025-04-23:
Same here - firmware is CPU 4.23.22, no connection possible anymore after update. Disabled "Auto Detect Additional Entities" now, then completely shut down the inverter (including battery and circuit breaker). After powering everything up again, it works - let's see if it stays that way.

WillCodeForCats, 2025-04-23:
This sounds like it might be an firmware issue if everyone is starting to have connection problems after updating the inverter.

lpersoons, 2025-04-23:
Probably same here. I also get a connection error since yesterday. Inverter CPU firmware is same as reported by other users 4.23.22. Disabling auto detect of other entities and restarting HA and inverter internet connection doesn't help. Will try other suggestions when I'm back from holidays.

thomas-ruw, 2025-04-24:
- I used the procedure from @danymexi yesterday evening. 
- disabled Auto Detect Additional Entities
- and reduced the polling interval from 10s to 30s

Up to now everything is still up and running. Let's keep the fingers crossed

I checked my logs and the modbus connection stopped right after the update to 4.23.22

AMichaelSuess, 2025-04-25:
Short update after two days of running without "Auto Detect Additional Entities" after a reset as described in my previous post: everything still works without issues and further intervention. Fingers crossed it stays that way.

sgbaker-hub, 2025-04-28:
Previously running 0004.0019.0039   Update pushed from somwehere (SE? , my installer?) and loaded 0004.0023.0027
Modbus comms lost as other posters noted.
Full inverter reset after disabling "Auto Detect Additional Entities" and now running without issues so far.
Note version of software I now have appears to be 4.23.27. not 4.23.22 as other posters.

thomas-ruw, 2025-04-28:
> * I used the procedure from [@danymexi](https://github.com/danymexi) yesterday evening.
> * disabled Auto Detect Additional Entities
> * and reduced the polling interval from 10s to 30s
>
> Up to now [...]

With the above changes it is running for 4 days now without a problem.
I'll revert the change in the polling interval, since disabling Auto Detect Additional Entities helped for the others.

Kishi85, 2025-04-28:
I'm getting the same issues now that SE decided to update one of my inverters to 4.23.27 since 2 hours ago (so they do work on stuff it seems as it is an even newer version than people have reported here so far) and just turning off "Auto-Detect Additional Entities" seems to fix the issue without any other steps necessary (at least for me). Other Apps (like evcc) using modbus to get data from the systems seem to work fine so far. 

From what I've learned with scripting Solaredge modbus directly on a RasPi for export limitation (since they cannot do this in their RRCR interface directly) this might be related to trying to read a register that is unavailable now on 4.23.xx which causes the inverter to kill the connection (using a modbus proxy this would kill all of them). I've not had the chance to check which register might be affected by manually iterating over them (as it seems the update is still ongoing atm). I'll add an update as soon as I can get more information (also will be contacting my installer tomorrow to see if they know anything as they have usually very good direct connections to people at SE).

Kishi85, 2025-04-28:
A small update: Now that the update to 4.23.27 seems to have finally finished and settled (at least on my secondary inverter now). I can once again query the inverter like before on 4.22.44 (with additional entities enabled). Everything seems to work as it should. 

My primary inverter seems to be on 4.22.44 still (I'm assuming they do a staged rollout) but all values are properly accessible on the inverter with 4.23.27 so it might have been 4.23.22 that was broken and they did a quick fix after the issue got known (as this interface is necessary for other commericial solutions to work like SolarLog for power control which SE even certified for use with their products).

Also since disabling "Auto-Detect Additional Entities" seems to fix the issue on 4.23.22 the problem must have been related to one of the values that is queried when enabling this. Those being RRCR Status/Active Powerlimit/CosPhi which might be/have been broken or did change in the way it could be queried (since Solaredge-Modbus-Multi retrieves them in a single read) and that caused the SE internal connection cleanup on inaccessible registers/bad modbus requests.

WillCodeForCats, 2025-04-28:
SolarEdge appears to have changed the behavior of unsupported modbus registers. Previously, it would return a modbus code for unsupported. Now it's hanging with no response. The detection routine relied on the modbus unsupported code, so I'll have to add something to detect timeouts.

tcjw3, 2025-05-02:
I also have the same problem and have not been able to read out any data via Modbus-tcp for 5 days. I have tried all the tips given here. The tip from @thomas-ruw helped me: Switch Modbus-tcp off once and then on again. It was not necessary to restart the inverter. 
However, typical beginner's mistake of impatient people, I also set the Modbus termination setting in RS485 meter 1 to ‘on’ at the same time. Now I can't say exactly which of the two got the system up and running again.
Both ‘SolarEdge Modbus Multi’ in HA and the Solaredge-app work perfectly.