WillCodeForCats/solaredge-modbus-multi #812: Add Number of Optimisers (reporting/not reporting)

skellycode, 2025-05-02:
**Describe the feature**
I propose adding two new Home Assistant entities:
  -one showing the total number of optimisers in the system
  -one showing the number of successfully reporting optimisers

**Screenshots**
N/A

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2025.03.0
 - solaredge-modbus-multi Version: 3.1.0

**Additional context**
Occasionally I need to re-pair my optimisers (usually after a grid power outage) as not all of them function correctly when the power comes back on.  This usually involves me having to notice that the solar isn't producing as much as expected, going out to the inverter, connecting to the wifi and waiting for it to load.
With this feature request implemented, it would be trivial to display optimiser status on a Home Assistant dashboard, and send a push notification in the event an optimiser isn't functioning correctly.
This would save me (and I presume, others as well) from days of lost generation, and time wasted going out to the inverter to check the status of the optimisers.

WillCodeForCats, 2025-05-02:
Optimizer data is not available over modbus.

skellycode, 2025-05-02:
A shame.  Oh well, nothing to do here!
Thanks mate

closed 2025-05-02 by skellycode