WillCodeForCats/solaredge-modbus-multi #815: Battery available energy exceeds rated energy

tieskuh, 2025-05-08:
### Describe the bug

I don't know for sure if it's a bug or if I misconfigured something.
My log is full of these warnings. Multiple per minute.

All reading seem to be correct.
I have 2 inverters, both with batteries.
The warnings are only about I1B1. No warnings about I2.

If it is just a warning, it might be better to throw them less often.

```
2025-05-08 08:51:45.964 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:51:53.202 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:52:01.200 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:52:09.139 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:52:18.305 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:52:30.097 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:52:38.177 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:52:46.484 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:52:54.404 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:53:02.351 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:53:10.461 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:53:18.266 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:53:30.209 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:53:38.349 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:53:46.482 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:53:54.250 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:54:02.679 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:54:10.666 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:54:18.235 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:54:29.666 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:54:37.682 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:54:45.413 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:54:53.110 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:55:01.346 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:55:09.553 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:55:17.350 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:55:28.241 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:55:36.500 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:55:44.089 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:55:52.649 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:56:00.023 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:56:07.301 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:56:15.619 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:56:23.858 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:56:31.824 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:56:39.264 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:56:47.792 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
2025-05-08 08:56:59.151 WARNING (MainThread) [custom_components.solaredge_modbus_multi.sensor] I1B1: Battery available energy exceeds rated energy. Set configuration for Battery Rating Adjustment when necessary.
[... cut ...]
### Home Assistant Version

2025.5

### solaredge-modbus-multi Version

3.1.2

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-05-08:
I'll update it to log once.

It's this setting:
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration#battery-rating-adjustment

tieskuh, 2025-05-08:
I misunderstood the setting. The Dutch translation a bit ambiguous. It translated to something like 'Battery power adjustment'. Which seems like you are adjusting the values. I should have read the wiki. Sorry.
But anyway, it might indeed be better to log it once so I hope my issue wasn't for nothing ;)
Thanks a lot!

closed 2025-05-27 by WillCodeForCats