WillCodeForCats/solaredge-modbus-multi #818: No connection Modbus to inverter

KStechnics, 2025-05-14:
### Describe the bug

Instellen mislukt, wordt opnieuw geprobeerd: Setup failed: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request.
After laterst update not possible to connect to the inverter. Home assistant is up to date and also both of the inverters. 2x 5kw hd wave inverters with both 1 home battery. 
Before the update everything was going perfect.
If you need more information please contact me.

### Expected behavior

Try to find how it comes that the connection isn't possible after updating.

### Diagnostic File

[home-assistant_solaredge_modbus_multi_2025-05-14T09-11-19.058Z.log](https://github.com/user-attachments/files/20204221/home-assistant_solaredge_modbus_multi_2025-05-14T09-11-19.058Z.log)

### Debug logs

```shell

```

### Home Assistant Version

Core 2025.5.1 Supervisor 2025.05.1 Operating System 15.2 Frontend 20250509.0

### solaredge-modbus-multi Version

3.1.2

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

IngmarStein, 2025-05-15:
+1. I have the same issue after updating to firmware 4.23.27 (https://www.solaredge.com/en/support/setapp-inverters-firmware) yesterday.

IngmarStein, 2025-05-15:
I was able to restore Modbus TCP by disabling and re-enabling it on the inverter.

KStechnics, 2025-05-15:
> I was able to restore Modbus TCP by disabling and re-enabling it on the inverter.

Ooh i tried it also and it worked. Thx for the help )

closed 2025-05-15 by KStechnics