WillCodeForCats/solaredge-modbus-multi #827: Active Power unavailable on follower inverter

johngianni, 2025-05-30:
### Describe the bug

Dear Dev,
first of all let me thank you for your great work.

I'm now testing the very latest pre-rel 2.0 version, as the current release has given me problems with the set_value command (it fails very often). With this latest pre-release the set value command is rock solid and I can change the Active Power of my inverters.
I'm on the latest version of HA.

I have two SetApp Inverters. The master is on the latest Firmware and all is working perfectly. The second (the follower) is on 0004.0022.0044 firmware. (it may be upgrade on it's own shortly. 

The problem is that after some minutes since reloading the integration or starting Home Assistant, the Active Power Sensor report as Unavailable, thus also the relative combo and sliders. 

[image]

Is it a problem with the firmware or could it be a bug in the latest code?

Thank you

### Expected behavior

Should Be reporting the value of the sensor as usual

### Diagnostic File

[config_entry-solaredge_modbus_multi-01JP26M88ESF6WPATHNE5D6XM4.json](https://github.com/user-attachments/files/20521717/config_entry-solaredge_modbus_multi-01JP26M88ESF6WPATHNE5D6XM4.json)

### Debug logs

```shell

```

### Home Assistant Version

2025.5.3

### solaredge-modbus-multi Version

3.1.3-pre.2

### Installation Type

HAOS

### Read the Instructions

- [x] My configuration follows https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration
- [x] I have read the Known Issues https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues
- [x] This issue is not about a template error (do not report template errors, read https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes)

WillCodeForCats, 2025-05-30:
Pre [Release v3.1.3-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v3.1.3-pre.3) changes the timeout restore behavior.

johngianni, 2025-05-31:
I confirm it's fixed. Thank you !!

johngianni, 2025-06-03:
I've spoken too soon. It's still happening, after hours rather then minutes, but it's happening.

WillCodeForCats, 2025-06-03:
That is how it's supposed to work. If it times out they will go unavailable and reappear after they respond again.

johngianni, 2025-06-03:
I see, thanks. It's indeed coming and going.

closed 2025-06-03 by johngianni

WillCodeForCats, 2025-06-03:
I unfortunately can't fix the inverter firmware. I can only change how the integration handles it. If the inverter stops responding to some requests I can try to work around those with timeouts or error handling so all the other data continues to work, but a permanent fix for the response issues would have to come from SolarEdge.

There are other reports of firmware updates causing problems here: https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/813